dji-sdk/Payload-SDK #219: E_DjiFlightControllerEmergencyStopMotor Clarification

Ellislee1, 2024-10-29:
Looking at the function here: [https://github.com/dji-sdk/Payload-SDK/blob/159708e6ef8920f41dce11ce666a3446e91481f5/psdk_lib/include/dji_flight_controller.h#L472](https://github.com/dji-sdk/Payload-SDK/blob/159708e6ef8920f41dce11ce666a3446e91481f5/psdk_lib/include/dji_flight_controller.h#L472)
```C++
/**
 * @brief Emergency stop motor in any case.
 * @note If you want to turn on motor after emergency stopping motor, you need to use the interface to send disable
 * command to quit lock-motor status.
 * @param cmd: see reference of E_DjiFlightControllerEmergencyStopMotor
 * @param debugMsg:inject debug message to flight control FW for logging, size limit: 10 bytes
 * @return Execution result.
 */
T_DjiReturnCode DjiFlightController_EmergencyStopMotor(E_DjiFlightControllerEmergencyStopMotor cmd,
                                                       char debugMsg[EMERGENCY_STOP_MOTOR_MSG_MAX_LENGTH]);
```

The description seems unclear. Once the `E_DjiFlightControllerEmergencyStopMotor` command has been issues how are the motors then reactivated? Is it possible to get some sample code that demonstrates this re-enabling?

dji-dev, 2024-10-30:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/120568'>Zendesk ticket #120568</a>:<br><br>Hello, we are sorry. We have confirmed that if this interface is called, the flight cannot be unlocked again. You can only restart the drone and then call the flight control function again to control the flight.
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-10-30:
Thank you. This is what we expected.

closed 2024-10-30 by Ellislee1