dji-sdk/Payload-SDK #223: 视频流格式问题如何影响视频流传输

fighting2020, 2024-11-19:
你好，希望尽快得到你的回复。
背景：M350 RTK，PSDK版本为3.9.2. PSDK运行在我的板子上。
对于这两种视频流格式：
DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT
DJI_CAMERA_VIDEO_STREAM_TYPE_H264_DJI_FORMAT
在目前版本的示例程序中，在test_payload_cam_emu_base.c文件中，对格式是设置是这样的：
    **returnCode = DjiPayloadCamera_SetVideoStreamType(DJI_CAMERA_VIDEO_STREAM_TYPE_H264_DJI_FORMAT);**
    if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        USER_LOG_ERROR("DJI camera set video stream error.");
        return DJI_ERROR_SYSTEM_MODULE_CODE_UNKNOWN;
    }
    **s_cameraVideoStreamType = DJI_CAMERA_VIDEO_STREAM_TYPE_H264_DJI_FORMAT;**
这样运行后，在Mobile SDK(遥控器)上观看示例的PSDK_0005.h264视频文件没有问题。
（1） 当我决定修改video stream type时，我将上述代码改成：
      returnCode = DjiPayloadCamera_SetVideoStreamType(DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT);
      s_cameraVideoStreamType = DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT;
这样编译运行的结果是：
[21.555][cam]-[Error]-[DjiPayloadCamera_SetVideoStreamType:799) remove negotiated device sync error: 0x000000E1.
[21.555][user]-[Error]-[DjiTest_CameraEmuBaseStartService:1433) DJI camera set video stream error.
[21.555][user]-[Error]-[main:311) camera emu common init error
这个问题，我在remove negotiated device sync error #178 看到有他人问过，当时回复说在处理，这是在3.9.2版本依然没有处理好是吗？
（2） 既然直接 DjiPayloadCamera_SetVideoStreamType不行，那么根据默认情况下，默认的视频格式就是DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT，所以，上述代码我注释掉，只留下一行有效，即：
//returnCode = DjiPayloadCamera_SetVideoStreamType(DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT);
s_cameraVideoStreamType = DJI_CAMERA_VIDEO_STREAM_TYPE_H264_CUSTOM_FORMAT;
这样编译运行后，没有报错，但Mobile SDK(遥控器)上显示黑屏，什么也没有，所以想问一下，我还能怎么处理使得我可以在Custom-H264格式下推流看是视频效果呢？

我的真实目的是从一个连接在板子上的USB相机获取H264视频流并实时推送至Mobile SDK(遥控器)上观看，显然，我取到的视频流格式一般而言是Custom-H264格式。所以，我想先从视频文件先测试在Custom-H264格式下视频可以顺利传输。
我注意到PSDK3.9.2的发布日志里写到：
[image]
按照这个说法，3.9.2应该可以传输Custom-H264格式视频流了，但是怎么做呢，多谢你的回复。

fighting2020, 2024-11-19:
> 3.9.2 is not supported CUSTOM_FORMAT. It is serious issue for me too. 3.7 works well in my enviroment.

thanks bro. 
In the #178 issue you raised, the officical reply suggested use version 3.8.1 as a compromise (does v3.8.1 work for you? ), i wonder if it works for me, i'll give it a try. If that still fails, maybe i'll further downgrade the version to see what would happen and hope the wizards in DJI will fix it asap.

akgongon, 2024-11-19:
#177 is the reason why I recommend 3.7 to you. It is not fatal.

fighting2020, 2024-11-21:
> #177 is the reason why I recommend 3.7 to you. It is not fatal.

OK，I checked #177 and also tested V3.8.1, 3.8.1 works for me by now, as a stopgap solution.

uzgit, 2025-09-26:
@LaVoute974 Can you give details on what specifically you had to change?

LaVoute974, 2025-09-30:
Hello @uzgit 
I almost did the same as you, but I'm using your code to stream the desktop, but it doesn't work, the screen turns grey randomly.
You can find my code here : https://github.com/dji-sdk/Payload-SDK/issues/297