dji-sdk/Payload-SDK #224: DJI_FC_SUBSCRIPTION_TOPIC_BATTERY_SINGLE_INFO_INDEX1 invalid timestamp in callback

Ellislee1, 2024-11-29:
We have found a bug with setting up a callback for the topic `DJI_FC_SUBSCRIPTION_TOPIC_BATTERY_SINGLE_INFO_INDEX1`
When testing the callback function `time->milliseconds` always returns a constant value after armed (same a microseconds) where as the other callbacks increase correctly.

For reference this is the callback function:

```C++
T_DjiReturnCode FCSubscription::Battery_Callback(
    const uint8_t* data, uint16_t dataSize,
    const T_DjiDataTimestamp* timestamp) {
  const auto* batteryData = reinterpret_cast<const T_DjiFcSubscriptionSingleBatteryInfo*>(data);

  std::cout << "ESC Timestamp: " << timestamp->millisecond << ", "<< timestamp->microsecond << std::endl;

  return DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS;
}
```

And this is how we are creating the callback:
```C++
  USER_LOG_INFO("Initializing subscription DJI_FC_SUBSCRIPTION_TOPIC_BATTERY_SINGLE_INFO_INDEX1...");
  returnCode = DjiFcSubscription_SubscribeTopic(
      DJI_FC_SUBSCRIPTION_TOPIC_BATTERY_SINGLE_INFO_INDEX1,
      DJI_DATA_SUBSCRIPTION_TOPIC_5_HZ, FCSubscription::Battery_Callback);

  if (returnCode != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
    USER_LOG_ERROR("subscription battery module error. %d", returnCode);
    throw std::runtime_error("subscription battery module error.");
  } else {
    USER_LOG_INFO("subscription battery module OK");
  }

  return DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS;
  ```

We believe this is a bug with the PSDK and not our code.

dji-dev, 2024-12-03:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123224'>Zendesk ticket #123224</a>:<br><br>Hello, thank you for your feedback. May I ask what version of drone firmware and PSDK you are using?
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-12-03:
We are using the latest for both

dji-dev, 2024-12-04:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123224'>Zendesk ticket #123224</a>:<br><br>Hello, thank you for your feedback. Are you referring to this? (Attachment). We are using the M350 model.<br><a href='https://djisdksupport.zendesk.com/attachments/token/ocLZTXfZCpF9oQlPd2VNIbcWs' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

Ellislee1, 2024-12-04:
I can't see the image. But we are using the M3D.

dji-dev, 2024-12-05:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123224'>Zendesk ticket #123224</a>:<br><br>Hello, thank you for the additional information. Our phenomenon is that our "timestamp" has been fixed to a certain value and it will not be updated.
<br><p><em><small>°°°</small></em></p>