dji-sdk/Payload-SDK #226: Auto start code

GregCornis, 2024-12-02:
In your documentation, it says that 
> After binding the SkyPort V2 with the sample program through DJI Assistant 2, when the payload device is mounted on the drone, the payload device will automatically run the sample program compiled or burned by the developer.

I don't see the program running automatically on a Raspberry Pi. The program works fine when I start it manually via SSH, but is not started by DJI. Did I misunderstand something ? How is the drone able to autostart the program ?

dji-dev, 2024-12-03:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123375'>Zendesk ticket #123375</a>:<br><br>Hello, I'm sorry, this is an error in the document content. We have reported it to the document team and asked them to correct it as soon as possible. After binding, you still need to run the PSDK program manually. It will not run the PSDK program by itself, but you can set the boot-up function on the development board to let it start by itself.
<br><p><em><small>°°°</small></em></p>

GregCornis, 2024-12-03:
Alright that' what I thought. Thanks!

closed 2024-12-03 by GregCornis

dji-dev, 2024-12-03:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123375'>Zendesk ticket #123375</a>:<br><br>Thank you for your understanding. You're welcome. If you have any further questions, please contact us again.
<br><p><em><small>°°°</small></em></p>