dji-sdk/Payload-SDK #227: M3E zoom issues

xenogit, 2024-12-03:
I am trying to perform pan tilt zoom on the stock camera provided.
I am able to pan and tilt, but when I tried zoom, nothing happens.

I tried using the DjiCameraManager example provided.

DjiCameraManager_Init() return error code 0.
DjiCameraManager_GetCameraType returns error 0. Camera type is DJI_CAMERA_M3E
DjiCameraManager_GetFirmwareVersion returns error code 0. Firmware version is 11.8.2.2

I am selecting E_DJI_TEST_CAMERA_MANAGER_SAMPLE_SELECT_SET_CAMERA_ZOOM_PARAM example.

Both the Optical Zoom and continuous zoom returns error code 0.

Not sure if I am using the correct zoom command or the camera is faulty.

Please help!

dji-dev, 2024-12-04:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123501'>Zendesk ticket #123501</a>:<br><br>Hello, we apologize for the inconvenience. Could you please tell us what is the firmware version and PSDK version of the M3E drone you are using?
<br><p><em><small>°°°</small></em></p>

xenogit, 2024-12-10:
From DJI Assistant 2:
firmware version is 10.01.0503

PSDK version is 3.9.2

dji-dev, 2024-12-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123501'>Zendesk ticket #123501</a>:<br><br>Hello, thank you for your additional information. We have verified this version and firmware, and the behavior is normal. Can you use the Pilot APP to operate the zoom successfully? This can also determine whether the aircraft function is abnormal. At the same time, you can provide us with the PSDK Debug log and we will help you further confirm it.https://sdk-forum.dji.net/hc/en-us/articles/38861383793945-How-to-export-PSDK-Debug-log
<br><p><em><small>°°°</small></em></p>

xenogit, 2025-01-17:
Thanks.

Double checked with the DJI Controller.
It seems that I was viewing the WIDE lens instead of the ZOOM lens in the front mounted camera.

closed 2025-01-17 by xenogit