dji-sdk/Payload-SDK #228: Jetson Orin NX USB hulk failed

blackjacket996, 2024-12-04:
按照官方网站关于[jetson nano采用E-port接口的方案：](https://developer.dji.com/doc/payload-sdk-tutorial/cn/quick-start/quick-guide/jetson-nano.html)
进行Jetson Orin NX通信链路搭建，在USB Hulk步骤失败
[image]

blackjacket996, 2024-12-04:
补充:除/opt/nvidia/l4t-usb-device-mode-start.sh文件在Orin NX Ubuntu22.04中存放路径为/opt/nvidia/l4t-usb-device-mode/nv-l4t-usb-device-mode-start.sh外, 其余操作均与官方示例一致

blackjacket996, 2024-12-04:
补充：在手动运行nv-l4t-usb-mode-start.sh时报错：
[image]

以下是Jetson Orin NX和大疆M300RTK之间的硬件连接，我用TypeC-OTG以及USB-USB线连接，没用采用USB-TypeC方法：
通过ls /dev/tty*可以看到两个串口ttyUSB0以及ttyACM0：
[image]

dji-dev, 2024-12-04:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123530'>Zendesk ticket #123530</a>:<br><br>您好，如果是M300机型，不需要配置BULK环境了，连接后检查端口生成没问题可以直接使用。
<br><p><em><small>°°°</small></em></p>

blackjacket996, 2024-12-05:
> Agent comment from Leon in [Zendesk ticket #123530](https://djisdksupport.zendesk.com/agent/tickets/123530):您好，如果是M300机型，不需要配置BULK环境了，连接后检查端口生成没问题可以直接使用。
> 
> _°°°_

> Agent comment from Leon in [Zendesk ticket #123530](https://djisdksupport.zendesk.com/agent/tickets/123530):您好，如果是M300机型，不需要配置BULK环境了，连接后检查端口生成没问题可以直接使用。
> 
> _°°°_

您好，感谢您的回复。如果是直接使用的话，那么PSDK头文件中的CONFIG_HARDWARE_CONNECTION应该是DJI_USE_UART_AND_USB_BULK_DEVICE还是 DJI_USE_ONLY_UART呢？如果是DJI_USE_ONLY_UART那么后续读取相机视频流是否会出问题

blackjacket996, 2024-12-05:
> Agent comment from Leon in [Zendesk ticket #123530](https://djisdksupport.zendesk.com/agent/tickets/123530):您好，如果是M300机型，不需要配置BULK环境了，连接后检查端口生成没问题可以直接使用。
> 
> _°°°_

在硬件连接后不开启USB HULK功能，运行demo会报错[issue](https://github.com/dji-sdk/Payload-SDK/issues/147#issue-2160845614)

blackjacket996, 2024-12-05:
> > Agent comment from Leon in [Zendesk ticket #123530](https://djisdksupport.zendesk.com/agent/tickets/123530):您好，如果是M300机型，不需要配置BULK环境了，连接后检查端口生成没问题可以直接使用。
> > _°°°_
> 
> 在硬件连接后不开启USB HULK功能，运行demo会报错[issue](https://github.com/dji-sdk/Payload-SDK/issues/147#issue-2160845614)

这个问题我已经解决，现在demo可以运行，但是在运行时会出现Gimbal的以下报错:
[image]

shiner-chen, 2025-01-21:
> 补充：在手动运行nv-l4t-usb-mode-start.sh时报错： [image]
> 
> 以下是Jetson Orin NX和大疆M300RTK之间的硬件连接，我用TypeC-OTG以及USB-USB线连接，没用采用USB-TypeC方法： 通过ls /dev/tty*可以看到两个串口ttyUSB0以及ttyACM0： [image]

请问你是怎么配置的，我在我的orin agx的板上，串口名称是ttyGS0和ttyUSB0，并没有ttyACM0这个串口。

blackjacket996, 2025-01-22:
> > 补充：在手动运行nv-l4t-usb-mode-start.sh时报错： [image]
> > 以下是Jetson Orin NX和大疆M300RTK之间的硬件连接，我用TypeC-OTG以及USB-USB线连接，没用采用USB-TypeC方法： 通过ls /dev/tty*可以看到两个串口ttyUSB0以及ttyACM0： [image]
> 
> 请问你是怎么配置的，我在我的orin agx的板上，串口名称是ttyGS0和ttyUSB0，并没有ttyACM0这个串口。

更换OTG-USB的线试一下，图中的连接方式是我测试后发现能找到ACM0的唯一方案

cchester25, 2025-05-06:
您好，您的问题得到解决了没，我也是用的orin NX，那个BULK影响您后续使用么，我不知道咋开启这个。

blackjacket996, 2025-05-06:
> 您好，您的问题得到解决了没，我也是用的orin NX，那个BULK影响您后续使用么，我不知道咋开启这个。

测试发现BULK不太需要，找到端口就行

cchester25, 2025-05-06:
> > 您好，您的问题得到解决了没，我也是用的orin NX，那个BULK影响您后续使用么，我不知道咋开启这个。
> 
> 测试发现BULK不太需要，找到端口就行

感谢您的帮助！不知道可否添加您的联系方式进行技术交流，我也在用M300RTK做开发，下面是我制作的payload。

https://github.com/hku-mars/FAST-LIVO2/issues/248

blackjacket996, 2025-05-06:
> > > 您好，您的问题得到解决了没，我也是用的orin NX，那个BULK影响您后续使用么，我不知道咋开启这个。
> > 
> > 
> > 测试发现BULK不太需要，找到端口就行
> 
> 感谢您的帮助！不知道可否添加您的联系方式进行技术交流，我也在用M300RTK做开发，下面是我制作的payload。
> 
> [hku-mars/FAST-LIVO2#248](https://github.com/hku-mars/FAST-LIVO2/issues/248)

你好，非常感谢邀请，本人最近已经不在研发相关内容，十分抱歉