dji-sdk/Payload-SDK #229: Problem setting up the bulk configuration using Jetson nano and M350

DPCW19, 2024-12-05:
Hello, 
I am trying to get the video feedback from my M350 dji drone using the e-port. I've followed the [dji documentation](https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html) without success. I am only able to get the telemetry using the serial port. I tried to set up the bulk port using the _2. Enable Jetson Nano's USB bulk function_ from the dji documentation and the README from the startup_bulk script. I did change the path `/home/dji/Desktop/startup_bulk/startup_bulk /dev/usb-ffs/bulk1 &` from `l4-usb-device-mode-start.sh` and `l4-usb-device-mode.sh` with my user.

When I run the command after reboot:
`ps -aux | grep startup_bulk` 
I only get the following output:
`my_user   12171  0.0  0.0   8944  2052 pts/6    S+   09:45   0:00 grep startup_bulk`

I also tried to add changes made by another user facing the same issue but with another drone : #179, but nothing has changed.

Thank you for your assistance.

dji-dev, 2024-12-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, is there any error when you run the script l4-usb-device-mode-start.sh manually? From the phenomenon, your script did not successfully generate the BULK node.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2024-12-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, thank you for your patience. We remember that the Jetson Nano device actually has a l4-usb-device-mode-start.sh file. You can check the /opt/nvidia* directory to see if there is a script file with the same name. You can copy it directly to the end of your script and start it up following your script (note that the path needs to be modified).

    enable_bulk=1if [${enable_bulk}-eq1]; thenmkdir-p /dev/usb-ffs    cfg_str="${cfg_str}+BULK1"mkdir-p /dev/usb-ffs/bulk1func=functions/ffs.bulk1mkdir-p"${func}"ln-sf"${func}""${cfg}"mount -omode=0777-ouid=2000-ogid=2000-t functionfs bulk1 /dev/usb-ffs/bulk1/home/kyle/work/DJI/startup_bulk/startup_bulk /dev/usb-ffs/bulk1 &sleep3cfg_str="${cfg_str}+BULK2"mkdir-p /dev/usb-ffs/bulk2func=functions/ffs.bulk2mkdir-p"${func}"ln-sf"${func}""${cfg}"mount -omode=0777-ouid=2000-ogid=2000-t functionfs bulk2 /dev/usb-ffs/bulk2/home/kyle/work/DJI/startup_bulk/startup_bulk /dev/usb-ffs/bulk2 &sleep3fimkdir -p "${cfg}/strings/0x409" echo "${cfg_str:1}" > "${cfg}/strings/0x409/configuration" udevadm settle -t 5 || : ls /sys/class/udc > UDC /sbin/brctl addbr pi4br0 /sbin/ifconfig pi4br0 ${net_ip} netmask ${net_mask} up if [${enable_rndis} -eq 1]; then /sbin/brctl addif pi4br0 usb0 /sbin/ifconfig usb0 down /sbin/ifconfig usb0 up fi
<br><p><em><small>°°°</small></em></p>

dji-dev, 2024-12-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, if you have modified the /opt/nvidia/l4t-usb-device-mode-start.sh file, you can directly restart the device without manually running the script, because some configuration statements will rely on permissions or resource usage, running during the device startup initialization phase can avoid many errors. You only need to pay attention to whether the corresponding services are running normally after startup (including the l4t-usb-device-mode-start.sh service, which can be viewed using systemctrl status **).
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-10:
Hello, 
I tried a lot of things, and I wanted to start clean so I flashed the OS back again into my jetson nano with Jetpack 6.1.

I don't have any file `l4t-usb-device-mode-start.sh` in `/opt/nvidia/` now.
I have the same setup as before (https://github.com/dji-sdk/Payload-SDK/issues/229#issuecomment-2523328549) and I can't fix it.
The error:

`nv-l4t-usb-device-mode-start.sh[839]: cp: cannot stat '/proc/device-tree/chosen/plugin-manager/': No such file or directory`

DPCW19, 2024-12-11:
Hello,

Thanks for the support !

Indeed, nv-l4t-usb-device-mode.service is active when removing the cp command. But notihng appears using lsusb. However, I can  see changes with `ip a` command:
Without usb connected : 
[image]

With usb connected :
[image]

Using psdk_wrapper ros2, I can start the service start_percepetion, but I dont get anything in the camera stream topic. I don't think that the perception is really up.

dji-dev, 2024-12-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, you can directly check the ps -ef|grep "startup" command to see if two BULK nodes are pulled up? Normally, you will get the following command:

    rsp@CUBE:~/workspace/E-Port/3.9.2/Payload-SDK-release-v3.9.2/build/bin$ ps -ef | grep "startup"root 678 1 0 17:42 ? 00:00:00 sshd: /usr/sbin/sshd -D [listener] 0 of 10-100 startupsroot 689 1 0 17:42 ? 00:00:00 /opt/startup_bulk/build/startup_bulk /dev/usb-ffs/bulk1root 729 1 0 17:42 ? 00:00:00 /opt/startup_bulk/build/startup_bulk /dev/usb-ffs/bulk2rsp 14853 1254 0 18:21 pts/0 00:00:00 grep --color=auto startup
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-11:
Hello,
Yes I do have /dev/usb-ffs/bulk1 and bulk2
[image]

dji-dev, 2024-12-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, if you have these two nodes, please configure the link as DJI_USE_UART_AND_USB_BULK_DEVICE in dji_sdk_config.h, and then refer to the configuration of this article: https://sdk-forum.dji.net/hc/zh-cn/articles/38622108040857-Linux%E5%B9%B3%E5%8F%B0%E9%85%8D%E7%BD%AEBULK
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-12:
Hello,

I already configured the dji_sdk_config.h, but I have some updates.

Now when I run the sample code from PSDK, I can go deeper into camera stream functions:
[
[image]
](url)

But then, nothing appear, I don't get anything else. I tried diffrent options and none of them output an image.
It's still better than before, because I don't get an USB bulk config error.

Does it implies that usb bulk is configured properly ?

dji-dev, 2024-12-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, have you installed LIBUSB correctly? And your opencv version, ffmpeg version, these will affect the implementation of this function. In addition, you can directly use the liveview function to verify, in fact, the effect is the same, that is, refer to the DjiTest_LiveviewRunSample function,
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-13:
Hello,

Yes everything is installed correctly.
The only difference between the documentation and my setup is that I have an USB-C connector on the Jetson side, not a micro-USB. Do you think that this could be problematic ?

dji-dev, 2024-12-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, could you upload the relevant hardware diagram so that we can better confirm for you that there may be confusion in the master-slave relationship of USB devices?
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-13:
Hello,

This is the only modification I have.
The E-Port is set to host.
[image]

dji-dev, 2024-12-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, we recommend that you use a TypeC to A cable and an OTG cable to connect. This will force your device to connect to the drone as a slave device, so there will be no problem with the master-slave relationship.
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-16:
Hello, 

Thanks for the feedback.

I am already using a TypeC to A cable and an OTG cable.

Is it normal that I cannot see any ports using lsusb ?

dji-dev, 2024-12-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, this is normal. lsusb enumerates the USB slave devices you are currently connected to. When your device is connected as a slave device, you cannot see the aircraft connected, unless your device is the host and the aircraft is the slave device.
<br><p><em><small>°°°</small></em></p>

DPCW19, 2024-12-17:
Hello,

Thanks for your help.

This is what I get when I run`./dji_sdk_demo_linux_cxx`

[image]

Do you see anything that prevents me from getting the perception, or fpv stream ?

dji-dev, 2024-12-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, it is normal so far. The PTZ function is not supported because your PSDK program is running on EXTENSION PORT. It is only supported when running on Payload Port. This error is as expected.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2024-12-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/123724'>Zendesk ticket #123724</a>:<br><br>Hello, we recommend that you check the version of opencv and ffmpeg. We are using: ffmpeg version 4.4 Copyright (c) 2000-2021 the FFmpeg developers, opencv-4.9.0
<br><p><em><small>°°°</small></em></p>

EugeneBogush, 2025-02-26:
speed 921600 ?

DPCW19, 2025-03-05:
Hello, 
Thanks for your help ! I chekced again, and the baudrate is correct.