dji-sdk/Payload-SDK #231: 连接FlyCart 30读取ttyUSB0出错 DjiAccessAdapter_Init:215

wqycg, 2024-12-19:
[image]

使用USB转TTL再转USB连接树莓派（Ubuntu 24.04），发现有`/dev/ttyUSB0`设备。
无人机固件已升级到最新。
同时设置串口的波特率为头文件里的460800。
`#define USER_BAUD_RATE              "460800"`
编译运行PSDK示例出现以下错误：
`[2.539][adapter]-[Info]-[DjiAccessAdapter_Init:215) Try identify UART0 connection failed. Probably because SDK adapter or aircraft not finish init or UART connect error.`

wqycg, 2024-12-19:
程序里是会强制改掉设备的波特率，貌似不需要自己设置。
然而这里没有看到FlyCart 30适用的波特率：
https://github.com/dji-sdk/Payload-SDK/blob/860751ae152f849454c26f1ed3a1c06382eb74ba/psdk_lib/include/dji_platform.h#L95

dji-dev, 2024-12-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/124875'>Zendesk ticket #124875</a>:<br><br>您好，建议您确认下USB-TTL的型号，当前FC30仅支持FT232和CP2102，如果使用两个USB-TTL，两个USB-TTL均需要使用这两种型号之一。
另外建议您重启FC30，待飞机上电以后，尽快启动PSDK程序，不排除是飞机倒计时把E-Port Lite通信断掉了。
<br><p><em><small>°°°</small></em></p>

wqycg, 2024-12-20:
保持连接状态重启无人机，可以了，谢谢。

closed 2024-12-20 by wqycg

dji-dev, 2024-12-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/124875'>Zendesk ticket #124875</a>:<br><br>您好，不客气，您后续调试的时候也需要注意这个倒计时机制，上电后大概会保留20-30s的时间检测PSDK程序是否运行。
<br><p><em><small>°°°</small></em></p>

johnjin404, 2025-03-18:
你好，我最近也在FC30无人机上遇到相同的问题，我是使用电脑运行psdk程序测试的，与飞机端连接的usb转ttl用的`ft232`，但是一直报`Try identify UART0 connection failed.`，也在保持连接状态下尝试重启过飞机
下面是我的日志文件

[DJI_0024_20250318_11-39-43.log](https://github.com/user-attachments/files/19305749/DJI_0024_20250318_11-39-43.log)

18678771845, 2025-11-27:
> 你好，typeC母转USB母的线你从哪买的？我现在卡在买不到一端插进飞机，另一端接TTL转USB的线

[image]

18678771845, 2025-12-18:
> > > [@18678771845](https://github.com/18678771845) 您指的是图中的这个USB-C OTG线吗
> > > <img alt="Image" width="2000" height="2000" src="https://private-user-images.githubusercontent.com/25783139/519605492-c9fa7739-ac34-450e-b766-dd3860c45059.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQyMzQyNDAsIm5iZiI6MTc2NDIzMzk0MCwicGF0aCI6Ii8yNTc4MzEzOS81MTk2MDU0OTItYzlmYTc3MzktYWMzNC00NTBlLWI3NjYtZGQzODYwYzQ1MDU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTExMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMTI3VDA4NTkwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIzYzgzOGIwMDVmMTEwM2U0MWY1ZDc5ZWQ1ODU0M2ExYTlhMjIxODE3NDFkYjkzNWJiNmIwMTU2M2JlZmM5MTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NyKMgP_iywZQeraxJSK_u92neB9vSBNXZ7v6rntSm1I">
> > 
> > 
> > 对呀对呀 就是这个 我从网上找不到呢，我看你买的这个是山泽的，我去了他家旗舰店，他说没有
> 
> 应该是普通的type-c公转usb母吧，而不是母转母

找到原因了，我开错方向了......开的对侧插板口 哈哈

wqycg, 2026-02-03:
> Hello, I am using an E-Port Lite connected in OTG mode with a CP2102 module. This CP2102 is connected to another CP2102 module that is plugged into the USB port of my Raspberry Pi 5. However, I am unable to establish a connection and I am encountering the same error that you mentioned. Could you please share what solution worked for you?

Would you mind sharing a picture of the module connection?