dji-sdk/Payload-SDK #232: Connecting a Raspberry Pi 4 to a M300

Momofaiz, 2024-12-19:
Hello, I want to connect my Raspberry Pi 4 to my M300 RTK as a payload to control the camera.

So i checked the PSDK documentation, I found this : 

[image]

Is it normal that they use a PSDK adapter, to connect the RPi 4 to the OSDK port?

PS : I wanted to ask the question before purchasing the adapter board!

dji-dev, 2024-12-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/124988'>Zendesk ticket #124988</a>:<br><br>Hello, if you want to develop a camera device based on the Raspberry Pi development board, do not connect to the OSDK port, you need to connect to the Payload Port port, and the kit you choose should be the SkyportV2 or X-Port kit. You can refer to here: https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/drone-port.html
<br><p><em><small>°°°</small></em></p>

Momofaiz, 2024-12-20:
Thanks for your response!

I want to use the Raspberry Pi 4 for camera control without adding any camera. I just want to control the existing camera on the drone.

Do you recommand PSDK or OSDK?

dji-dev, 2024-12-20:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/124988'>Zendesk ticket #124988</a>:<br><br>Hello, sorry, I may have misunderstood you. If you want to control the camera, you need to connect to the E-Port/OSDK port instead of the Payload Port, and the kit used is the E-Port kit.
It is recommended that you use the PSDK program. Because the OSDK is no longer updated and maintained, the bugs contained in it will affect your function development and experience. In addition, please note that the SDK only supports DJI's official payload devices by default, and does not support the control of third-party payload camera devices.
<br><p><em><small>°°°</small></em></p>

Momofaiz, 2024-12-20:
Okay!

So, I should connect my Raspberry Pi 4 to OSDK Port, and I should use the PSDK program?

Knowing that I am going to use an official DJI Payload.

dji-dev, 2024-12-23:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/124988'>Zendesk ticket #124988</a>:<br><br>Hello, yes, you are right this time! Connect to the OSDK port and run the PSDK program. You can refer to the connection method here: https://sdk-forum.dji.net/hc/en-us/articles/34834641217305-PSDK-connection-of-various-aircraft-models
<br><p><em><small>°°°</small></em></p>

closed 2025-01-10 by Momofaiz