dji-sdk/Payload-SDK #233: FlyCart 30有没有办法获取毫秒精度的GPS时间

wqycg, 2024-12-21:
希望获取精度到0.1秒，也就是100毫秒的GPS时间。
消息订阅能够获取5Hz的GPS时间（`T_DjiFcSubscriptionGpsTime`），但只到秒，所以5Hz会存在最大0.2秒的时间误差。
FlyCart 30无法使用时间同步和精准定位功能，请问后续是否会支持？
请问有办法获取到0.1秒的GPS时间信息？

dji-dev, 2024-12-23:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/125087'>Zendesk ticket #125087</a>:<br><br>您好，很抱歉，目前暂时没有办法能拿到精度更高的GPS时间数据，这个主要是与飞控推送相关。方便问下您的需求和贵司的基本信息吗？我们和内部团队反馈下。
<br><p><em><small>°°°</small></em></p>

wqycg, 2024-12-23:
好吧，那我们先用这个了。

closed 2024-12-23 by wqycg

dji-dev, 2024-12-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/125087'>Zendesk ticket #125087</a>:<br><br>您好，很抱歉为您带来的不便，关于您的需求我们已记录，并已和内部团队反馈，后续还请您留意PSDK的发布公告，如果有新增对此功能的支持，我们也会列出来的。
<br><p><em><small>°°°</small></em></p>