dji-sdk/Payload-SDK #234: About perception image data id

XinnWang, 2025-01-08:
We are using the PSDK 3.9.2. Our paltform is M3E. For perception image, we subscribe only front direction channel. We print the image data id inside the callback function, and find this id is not always continuous, for example: ... 1000 1001 1002 1002 1004 ... , which means 1002th is repeated twice and 1003th frame is lost. Average shows for every 100 frames this lost happens once. Is this correct or anything wrong with our configurations?

dji-dev, 2025-01-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, I may trouble you to confirm whether it is 1 frame lost every 100 frames or once every 1000 frames. If it is every 100 frames, we may need to confirm the aircraft's status at the time through the log. Under normal circumstances, it is normal to lose one frame every 1000 frames. This is affected by the current resource usage of the aircraft. You do not need to do too much configuration for this function, just call the relevant interface.
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-08:
@dji-dev How can i get the log for the aircraft's status?

dji-dev, 2025-01-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, you can refer to here:How to export aircraft logs/parameter logs
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-09:
@dji-dev 
[image]
This is the aircraft logs. During this log, we subscribe only down direction stereo channel. We lost almost 20% left images, and 20% right images.

dji-dev, 2025-01-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, we are sorry for the bad experience. We have received your log. What is the probability of this problem happening to you? Or have you found a scenario where this problem must occur? We may be able to try to reproduce it by aligning your method.
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-09:
Just normal offlice indoor scenario, the aircraft is only power up but not take off, we take the aircraft by hand and move it in the air, the probability of this issue is almost 100%. We record the image id inside callback function, and analysis the numbers afterwards. It is easy to find there is lots of repeated and lost ids. For down direction stereo is almost 20%.

dji-dev, 2025-01-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, I understand. We will try to align your environment to reproduce the problem. Thank you for your supplement. Finally, we would like to confirm whether you still remember the time when the problem occurred. Regarding the log, if we use the problem time, we can locate the content more quickly.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-01-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, thank you for your additional information. We will analyze your log as soon as possible. In addition, our R&D team has information to share with you: it is suspected that the auxiliary impact function has affected the bandwidth resources of the aircraft. You can try to turn off the auxiliary impact function on the Pilot and obtain the grayscale image again to see if the effect is improved.
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-10:
"the auxiliary impact function" in chinese is "视觉避障功能“ ?

dji-dev, 2025-01-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, the Chinese version should be "飞行辅助影像".
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-14:
We close the auxiliary impact function, the lost rate drops to around 0.5%. But during the aircraft taking off, the lost rate increases to around 10% per channel. This shows the lost is getting worse when the aircraft is taking off.

dji-dev, 2025-01-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, we apologize for the inconvenience. Our R&D team is analyzing your drone logs. You can check whether the drone firmware is the latest version. To avoid the impact of known issues, we recommend that you develop and test based on the latest firmware. We will synchronize the conclusions for you as soon as possible after log analysis.
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-15:
When running the sdk demo, it prints a lot of this warning:
```
[linker]-[Warn]-[DjiProtocol_v1Unpack:187) protocol frame crc16 error, exp:0x35E7 cur:0x6369
```
Is this means something wrong in our configuration? Will this affect the data transfer performance?

dji-dev, 2025-01-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, this is an abnormal message related to the serial port. If it appears frequently, it means that there may be a problem with your serial port link (which will affect your data sending function). It is recommended that you change the baud rate or replace the serial port hardware link.
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-01-16:
After changing to a USBtoTTL device, there is no more warning, and the lost rate drops to 0.5% for front and 5% for down. so the lost is mainly due to the uart. For the uart port, we use the uart0 from orin nx which original is 1V8, we use a levelshifter to change to 3V3. Is there anything wrong with our uart configuration?

dji-dev, 2025-01-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126459'>Zendesk ticket #126459</a>:<br><br>Hello, according to your description, it seems to be a problem with the onboard serial port itself. The onboard serial port itself may have problems such as frequency deviation, which will directly affect the communication of your PSDK.
<br><p><em><small>°°°</small></em></p>