dji-sdk/Payload-SDK #236: Bad GPS Position

Momofaiz, 2025-01-10:
Hello,

I am developping a PSDK application using a Raspberry Pi 4 and a dji M300 RTK.

When I execute my code:

[image]

My application is based on the same code as The PSDK sample application for nvidia jetson.

But, when i execute the sample application, I get same data from the drone:

[image]

Here is the code responsible of the GPS data subscription:

```cpp
ordered_json Position::getGpsData() {
    T_DjiFcSubscriptionGpsPosition gpsPosition;
    T_DjiFcSubscriptionVelocity velocity;
    T_DjiDataTimestamp timestamp = {0};

    DjiFcSubscription_SubscribeTopic(DJI_FC_SUBSCRIPTION_TOPIC_GPS_POSITION, DJI_DATA_SUBSCRIPTION_TOPIC_1_HZ, NULL);
    DjiFcSubscription_SubscribeTopic(DJI_FC_SUBSCRIPTION_TOPIC_GPS_VELOCITY, DJI_DATA_SUBSCRIPTION_TOPIC_1_HZ, NULL);

    if (DjiFcSubscription_GetLatestValueOfTopic(DJI_FC_SUBSCRIPTION_TOPIC_GPS_POSITION, reinterpret_cast<uint8_t*>(&gpsPosition), sizeof(gpsPosition), &timestamp) != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        std::cerr << "Erreur : Impossible de récupérer les données GPS" << std::endl;
    }
    if (DjiFcSubscription_GetLatestValueOfTopic(DJI_FC_SUBSCRIPTION_TOPIC_GPS_VELOCITY, reinterpret_cast<uint8_t*>(&velocity), sizeof(velocity), &timestamp) != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        std::cerr << "Erreur : Impossible de récupérer les données de vitesse" << std::endl;
    }

    ordered_json drone_data = {
        {"latitude", gpsPosition.y},
        {"longitude", gpsPosition.x},
        {"altitude", gpsPosition.z},
        //{"satelliteCount", gpsPosition.gpsSatelliteNumber},
        {"velocity", {{"x", velocity.data.x}, {"y", velocity.data.y}, {"z", velocity.data.z}}}
    };

    return drone_data;
}
```

dji-dev, 2025-01-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126763'>Zendesk ticket #126763</a>:<br><br>Hello, when you subscribe to GPS information, did you turn on the simulator in the DJI Assistant 2 software?
<br><p><em><small>°°°</small></em></p>

Momofaiz, 2025-01-13:
Hello @dji-dev, Thanks for your response!

No, I don't turn on the simulator in the DJI Assistant 2. In fact, I don't use DJI Assistant 2 at all.

But, your PSDK sample application worked well without DJI Assistant 2.

dji-dev, 2025-01-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126763'>Zendesk ticket #126763</a>:<br><br>Hello, thank you for your supplement. Could you provide the debug level log of your PSDK? We will further check your issue.How to export PSDK Debug log
<br><p><em><small>°°°</small></em></p>

Momofaiz, 2025-01-14:
Hello @dji-dev, 

There is the logfile : 

[DJI_0008_20250114_12-37-01.txt](https://github.com/user-attachments/files/18409922/DJI_0008_20250114_12-37-01.txt)

dji-dev, 2025-01-15:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126763'>Zendesk ticket #126763</a>:<br><br>Hello, sorry, could you please confirm again whether your log file is the corresponding file? We checked your log and there is no corresponding content. (We are concerned about the operation you subscribed to)
<br><p><em><small>°°°</small></em></p>

Momofaiz, 2025-01-17:
Hello @dji-dev,

I have made a bad mistake. I Started my code with FC Subscription without FC init.

```cpp
if (DjiFlightController_Init(ridInfo) != DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS) {
        std::cerr << "Init flight controller failed.\n";
    }
```
That is why there was no Subscription log in the file.

Thanks a lot for your support!

dji-dev, 2025-01-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/126763'>Zendesk ticket #126763</a>:<br><br>Hello, I'm glad that you found and solved the problem. You're welcome. If you have any further questions, you are welcome to contact us again.
<br><p><em><small>°°°</small></em></p>