dji-sdk/Payload-SDK #237: Updating dji sdk policy file... 串口写失败

shiner-chen, 2025-01-17:
我想通过psdk在orin agx板上拉取M300 RTK无人机上的视频流，根据官网的操作指南，连接了硬件，在运行psdk的demo程序：dji_sdk_demo_on_jetson， 报以下错误，不知道如何处理？
```
[76.340][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:493) Updating dji sdk policy file... 
[76.356][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5239>Command async send retry: index = 10, retryTimes = 65517, 0x0A06->0x0701 0x3C40 
[76.356][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.356][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.356][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.360][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5237>Command async send retry: index = 8, retryTimes = 65507, 0x0A06->0x0701 0x3C40 
[76.360][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.360][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.360][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.360][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5238>Command async send retry: index = 9, retryTimes = 65512, 0x0A06->0x0701 0x3C40 
[76.360][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.360][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.360][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.367][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5236>Command async send retry: index = 7, retryTimes = 65502, 0x0A06->0x0701 0x3C40 
[76.367][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.367][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.367][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.373][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5234>Command async send retry: index = 5, retryTimes = 65492, 0x0A06->0x0701 0x3C40 
[76.373][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.373][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.373][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.376][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5235>Command async send retry: index = 6, retryTimes = 65497, 0x0A06->0x0701 0x3C40 
[76.376][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.376][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.376][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
[76.378][linker]-[Warn]-[DjiCommand_SendAsyncHandle:965) <0x5233>Command async send retry: index = 4, retryTimes = 65487, 0x0A06->0x0701 0x3C40 
[76.378][linker]-[Error]-[DjiChannel_uartWrite:102) uart write data error 
[76.378][linker]-[Error]-[DjiLinker_SendMsg:198) write data error 
[76.378][linker]-[Warn]-[DjiCommand_SendAsyncHandle:971) Command retry send failed, errorCode:0x000000EC 
```
硬件连接方式如下：

[image]

nv-l4t-usb-device-mode-start.sh脚本如下：
```
#!/bin/bash
set -ex

# Ensure the script runs as root
if [ "$(id -u)" -ne 0 ]; then
    echo "This script must be run as root"
    exit 1
fi

# Load required modules
sudo modprobe configfs
sudo modprobe libcomposite
sudo modprobe usb_f_fs

# Mount configfs if not already mounted
if ! mountpoint -q /sys/kernel/config; then
    sudo mount -t configfs none /sys/kernel/config
fi

script_dir="$(cd "$(dirname "$0")" && pwd)"
. "${script_dir}/nv-l4t-usb-device-mode-config.sh"

# Wait for any modules to load and initialize
for attempt in $(seq 60); do
    udc_dev_t210=700d0000.xudc
    if [ -e "/sys/class/udc/${udc_dev_t210}" ]; then
        udc_dev="${udc_dev_t210}"
        break
    fi
    udc_dev_t186=3550000.usb
    if [ -e "/sys/class/udc/${udc_dev_t186}" ]; then
        udc_dev="${udc_dev_t186}"
[... 263 lines cut ...]
```

根据lsusb -d 0955:7020 -v 输出的内容，将hal_usb_bulk.h 文件中bulk配置如下：
```

/* Exported constants --------------------------------------------------------*/
#define LINUX_USB_BULK1_EP_IN_FD "/dev/usb-ffs/bulk1/ep1"
#define LINUX_USB_BULK1_EP_OUT_FD "/dev/usb-ffs/bulk1/ep2"

#define LINUX_USB_BULK1_INTERFACE_NUM (7)
#define LINUX_USB_BULK1_END_POINT_IN (0x88)
#define LINUX_USB_BULK1_END_POINT_OUT (0x05)

#define LINUX_USB_BULK2_EP_IN_FD "/dev/usb-ffs/bulk2/ep1"
#define LINUX_USB_BULK2_EP_OUT_FD "/dev/usb-ffs/bulk2/ep2"

#define LINUX_USB_BULK2_INTERFACE_NUM (8)
#define LINUX_USB_BULK2_END_POINT_IN (0x89)
#define LINUX_USB_BULK2_END_POINT_OUT (0x06)

#define LINUX_USB_VID (0x0955)
#define LINUX_USB_PID (0x7020)

/* Exported types ------------------------------------------------------------*/
```

dji-dev, 2025-01-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127344'>Zendesk ticket #127344</a>:<br><br>您好，请问下您使用的是板载串口还是USB-TTL，这个主要是串口相关的报错，建议您更换下串口连接链路，或者调整一下波特率。同时您可以确认下无人机的固件版本是否为最新。
<br><p><em><small>°°°</small></em></p>

huozirong, 2025-09-02:
你好，请问你解决了吗，我也遇到了这个问题，运行psdk的时候一直卡在更新策略那里

daz, 2026-02-21:
I am getting the same error using Raspberry Pi sample from this repo, M300 using PDSK version 3.14.0