dji-sdk/Payload-SDK #238: Jetson Nano download image failed

wuxiangchao, 2025-01-17:
when I execute:
 ` sudo ./dji_sdk_demo_on_jetson_cxx`
choice:

[image]
and choice 1 next

[image]

next choice 11:

[image]

get the following output, how to resolve this error with timeout?

[image]

dji-dev, 2025-01-17:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127345'>Zendesk ticket #127345</a>:<br><br>Hello, timeout is usually caused by link problems. What link did you select in the dji_sdk_config.h file? Please make sure that the link is normal. In addition, you can try to download media files on the pilot to confirm whether the P1 camera functions normally.
<br><p><em><small>°°°</small></em></p>

wuxiangchao, 2025-01-19:
> Agent comment from Leon in [Zendesk ticket #127345](https://djisdksupport.zendesk.com/agent/tickets/127345):Hello, timeout is usually caused by link problems. What link did you select in the dji_sdk_config.h file? Please make sure that the link is normal. In addition, you can try to download media files on the pilot to confirm whether the P1 camera functions normally.
> 
> _°°°_

Thanks for your reply, I'm using jetson nano dev kit, and connect to osdk port (M300 RTK) with psdk, could I download the camera (P1) image from M300 RTK device?

wuxiangchao, 2025-02-07:
[image]