dji-sdk/Payload-SDK #241: Specify CameraSource when setting camera parameters

michele-colombo, 2025-01-22:
Hi,
we are using PSDK with DJI M300 RTK with H20 payload. We are trying to use `DjiCameraManager_SetFocusRingValue`, but we noted it only works if the selected camera source, on the receiver, is zoom and not wide. I know that the wide camera does not support focus (it has fixed focus), but we would like to be able to set the focus for the zoom camera, even if the current camera displayed on the receiver is wide. Would it be possible to add the possibility to specify the CameraSource (wide, zoom, ...) when calling `DjiCameraManager_SetFocusRingValue`, in a similar way to what is done with `DjiLiveview_StartH264Stream`? The signature would become something like: 
```c++
T_DjiReturnCode DjiCameraManager_SetFocusRingValue(E_DjiMountPosition position,
                                                   uint16_t value, E_DjiCameraSource source = 0);
```
Our use case is the following: we start the streaming of the wide camera to be used by a companion computer to perform navigation tasks; meanwhile we start the recording of the zoom camera, which will be the output of the drone inspection. Thus we would like to be able to stream the wide camera, but set the focus ring value on the zoom camera.
Thanks! 
Michele

dji-dev, 2025-01-23:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127867'>Zendesk ticket #127867</a>:<br><br>Hello, thank you for your patience. Regarding your requirements, have you tried to use multi-threading to achieve it? We noticed that you should want to get the wide-angle video stream while also setting the focus ring of the zoom camera.
<br><p><em><small>°°°</small></em></p>

michele-colombo, 2025-01-23:
Thanks for your reply!
My goal is exactly as you stated: received wide camera stream and set focus ring on zoom camera.
Could you explain in more detail how multi-threading could help achieve this? My understanding is that the issue arises because starting the H.264 stream on the wide camera causes subsequent requests—such as setting the focus ring—to be interpreted as targeting the streaming camera, leading to failures. This is why I was asking if there is any way to specify the camera source when sending a command or if will be possible in any next release.

dji-dev, 2025-01-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127867'>Zendesk ticket #127867</a>:<br><br>Hello, thank you for your supplement. We may need to try it locally first. If it is feasible, we will synchronize the solution to you. If it is not feasible, we will synchronize it to the R&D team to see if there is a good solution, or let them evaluate whether to add this function. Due to the approaching Spring Festival holiday, the response to this issue may not be too fast. After the holiday is over, we will contact you as soon as possible. Thank you again for your patience and we apologize for the inconvenience caused to you~
<br><p><em><small>°°°</small></em></p>

michele-colombo, 2025-01-24:
Thank you so much! I'm looking forward to hearing from you.

dji-dev, 2025-02-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127867'>Zendesk ticket #127867</a>:<br><br>Sorry to have kept you waiting for so long. We have tried the previously mentioned method locally, but it is not feasible. There is no way to switch or set the focal length in the background with a wide-angle lens. Regarding this requirement, we have helped you to feedback to the internal team for evaluation. Could you please briefly introduce the application scenario where you need this function?
<br><p><em><small>°°°</small></em></p>

michele-colombo, 2025-02-10:
In addition to what I wrote in the opening post, I can add the following to our scenario.
We are performing automated inspections of wind turbines. We use the wide camera for automated navigation of the drone (we process its output on a companion Jetson computer and compute the direction of movement to perform the inspection), while the zoom camera produces the actual output of the inspection: a high-resolution, close shot of the blade under inspection. In order to do this, we start streaming on the wide camera and recording on the zoom camera (actually on both, even though recording for wide is not strictly needed): this works fine. In order to have to best possible quality on the inspection video (from zoom camera), we want to set the focus on the zoom camera manually, by using the estimated distance from the blade under inspection. Note that we already tested setting the focus ring on zoom camera and this works fine when the wide camera is not streaming (and thus zoom camera is the active camera). Note that we cannot use a single camera (either zoom or wide) for both inspection and navigation, as they have opposite requriements: for navigation we need a wide FOV in order to compute a better navigation path, while for inspection we need an image as close as possible of the object under inspection. I hope this clears why we need to stream the wide camera and meanwhile set the focus on the zoom camera (which is recording).

I mentioned focus because that is our most stringent requirement right now, but it could be useful to set other parameters (such as exposure, aperture, etc.) on the zoom camera while the wide is streaming.

Thank you!

dji-dev, 2025-02-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127867'>Zendesk ticket #127867</a>:<br><br>Hello, thank you for your additional requirement scenario! We have understood your usage scenario and main issues, and fed back this information to the internal team. We will keep this ticket open until we update you with the evaluation opinions of the R&D team.
<br><p><em><small>°°°</small></em></p>

michele-colombo, 2025-03-13:
Hi, do you have any new on this? Thanks!

dji-dev, 2025-03-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/127867'>Zendesk ticket #127867</a>:<br><br>Hello, sorry to have kept you waiting. We have received information from the R&D team today. Regarding the evaluation of this requirement, the existing model software framework may not be supported and adapted. Unfortunately, this may not meet your needs. However, we will consider adapting this in advance on subsequent new models.
<br><p><em><small>°°°</small></em></p>