dji-sdk/Payload-SDK #243: Issue Connecting the Mavic3 with E-Port

JaMateoslu, 2025-02-04:
The E-Port is connected as well as a data communication cable from USB-C to USB to enable the PSDK but I am unable to see it on my Laptop.
I am running Ubuntu 22.04 with ROS2 Humble installed.

The drone is connected but it cannot be seen on any of the USB Ports when calling the following method:

javibuntu@javibuntu-Thin-GF63-12VE:~$ ls /dev/usb-ffs/
ls: no se puede acceder a '/dev/usb-ffs/': No existe el archivo o el directorio
javibuntu@javibuntu-Thin-GF63-12VE:~$ ls /dev/ttyACM*
ls: no se puede acceder a '/dev/ttyACM*': No existe el archivo o el directorio
javibuntu@javibuntu-Thin-GF63-12VE:~$ 

Only when listing the devices is when it appears but cannot be mapped as mentioned before, thus the warpper cannot be initialized.

[image]

When tryin to find the serial port I am unable to do so:
javibuntu@javibuntu-Thin-GF63-12VE:~$ ls /dev/ttyUSB* /dev/ttyACM* /dev/serial/by-id/*
ls: no se puede acceder a '/dev/ttyUSB*': No existe el archivo o el directorio
ls: no se puede acceder a '/dev/ttyACM*': No existe el archivo o el directorio
ls: no se puede acceder a '/dev/serial/by-id/*': No existe el archivo o el directorio

This is the hardware connection of the drone with the SDK according to the documentation.

Drone Mavic 3T --> E-Port + Computer 

Have tried with different cables USBC to USB and USBC

The E-Port Development Kit light turns greens but dissapears after a couple seconds unless I press it hard against the E-Port.

[image]

siviae, 2025-05-02:
I had the same issue, it turns out that you must connect E-port Development Kit to the drone before turning it on, not after. If you do it after, then it turns off power at e-port after a couple of seconds.

siviae, 2025-06-17:
> Hi, I encountered the same issue. DJI told me it's a problem with the drone's cable connection slot, apparently a design flaw.
> 
> They sent me a replacement cable, but it's still loose. Were you able to solve the problem?

My cable is also loose by the way, I plan to fix this issue using hot glue.

Y1annis, 2025-09-16:
Did anyone find a fix to the issue? I have the same problem, e-port starts and closes after a few seconds but when I push it into the drone it stays on