dji-sdk/Payload-SDK #244: PSDK lib on cortex-m4 gcc

NotNullDev, 2025-02-08:
Hello,

 I am quite new to the embeded world (that's why I really want to stick to the official board tool).

I am using official IDE (cubeIDE) for my mcu (STM32f405xx). The issue is that cubeIDE uses gcc not armcc.

I was able to install FreeRTOS on my mcu, however I am unable to import the PSDK library, because (forgive me if I am mistaken or this one) it was compiled using Keil/ARMCC.

According to the Chat GPT it is possible to create such library that has compatibility with the GCC by adding flag `--target=arm-arm-none-eabi` during compilation.

Would it be possible that you include library for the gcc version as well?

NotNullDev, 2025-02-09:
I've just found...

https://github.com/dji-sdk/Payload-SDK/commit/b96c87f4bb71169b3d67fc9a2fbabc2048851827#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5L39-R43

This commit also removed other libs from the github repository...

https://github.com/dji-sdk/Payload-SDK/blob/a8dc2f5628cc47ae490e2aa6ee8c16679e8bb91a/samples/sample_c/platform/rtos_freertos/stm32f4_discovery/project/armgcc/CMakeLists.txt#L58

There is still example using this non-existing library.

Guys, PLEASE add README.md to the folder with the libraries with the information that you offer in fact other sdks. I've just wasted a little bit of that because there is no mention about that in any visible (if any at all) place...

dji-dev, 2025-02-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/128663'>Zendesk ticket #128663</a>:<br><br>Hello, for any PSDK static library, you can first search here to see if it contains the static library file you need: https://pan-sec.djicorp.com/s/smKrWMrPRJbe7YJ
<br><p><em><small>°°°</small></em></p>

NotNullDev, 2025-02-10:
> Agent comment from Leon in [Zendesk ticket #128663](https://djisdksupport.zendesk.com/agent/tickets/128663):Hello, for any PSDK static library, you can first search here to see if it contains the static library file you need: https://pan-sec.djicorp.com/s/smKrWMrPRJbe7YJ
> 
> _°°°_

Thanks! This looks like exactly what I was looking for!

Would it be possible for you to add a README.md file with the link you provided, ideally in the psdk_lib directory? 

Alternatively, could you at least mention it in the existing README so others don’t waste time searching for it? 

I spent quite a few hours this weekend looking for available solutions. 😄

closed 2025-02-10 by NotNullDev