dji-sdk/Payload-SDK #245: 关于PSDK飞行控制的问题

XinnWang, 2025-02-10:
您好，我们准备在无GPS区域通过PSDK控制飞机飞行，我们看到文档中有这样一段描述：
```
Note: 只有当GPS信号良好（health_flag >=3）时，可使用与水平位置控制（DJI_FLIGHT_CONTROLLER_HORIZONTAL_POSITION_CONTROL_MODE）相关的控制模式。只有当GPS信号良好（health_flag >=3）或高级感应系统与Autopilot正常工作时，才可使用与水平速度控制（DJI_FLIGHT_CONTROLLER_HORIZONTAL_VELOCITY_CONTROL_MODE）相关的控制模式。
```
关于这段话有几个问题想咨询一下：
1. 对于无GPS区域如果我们自己有定位方案，可以使用水平位置控制么？
2. 高级感应系统和Autopilot是指的什么功能？对于无GPS区域是否可以使用水平速度控制？

dji-dev, 2025-02-10:
Agent comment from LIPING.ZHOU in <a href='https://djisdksupport.zendesk.com/agent/tickets/128728'>Zendesk ticket #128728</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

1.无GPS环境下，仅推荐使用油门、姿态角控制无人机，否则因为无GPS会影响位置和速度控制，以下是相关文档：5.7 joystick飞行控制（flight_control）
2.同2 ，Autopilot您指的 应该是航点功能吧，您可以查阅以下文档了解下：5.18 航点任务（waypoint_mission）

希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-02-10:
Agent comment from LIPING.ZHOU in <a href='https://djisdksupport.zendesk.com/agent/tickets/128728'>Zendesk ticket #128728</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

不建议无GPS使用未推荐的其他模式，可能会有炸机风险。

希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-02-11:
您好，我在论坛看到避障功能只在速度控制模式下生效，请问最新版本PSDK也是这个状态么？

dji-dev, 2025-02-11:
Agent comment from LIPING.ZHOU in <a href='https://djisdksupport.zendesk.com/agent/tickets/128728'>Zendesk ticket #128728</a>:<br><br>尊敬的开发者，

您好，感谢您联系DJI 大疆创新。

是的，只有速度控制模式有避障

希望我们的解决方案能够帮到您，感谢您的邮件，祝您生活愉快！

Best Regards,
DJI 大疆创新SDK技术支持
<br><p><em><small>°°°</small></em></p>

XinnWang, 2025-02-11:
请问速度控制模式下可以绕障么？

closed 2025-02-13 by XinnWang