dji-sdk/Payload-SDK #246: Missing arm-none-eabi-gcc library

QE-Nicholas, 2025-02-12:
Hi, the old version PSDK comes with arm-none-eabi-gcc library while the new version is missing it.

dji-dev, 2025-02-12:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/128915'>Zendesk ticket #128915</a>:<br><br>Hello, sorry for the inconvenience, you can download it from here: https://pan-sec.djicorp.com/s/smKrWMrPRJbe7YJ
<br><p><em><small>°°°</small></em></p>

yushikong, 2025-03-07:
> Agent comment from Leon in [Zendesk ticket #128915](https://djisdksupport.zendesk.com/agent/tickets/128915):Hello, sorry for the inconvenience, you can download it from here: https://pan-sec.djicorp.com/s/smKrWMrPRJbe7YJ
> 
> _°°°_
能否尽快把3.11的社区版本静态库发布一下？
建议把arm-none-eabi-gcc的静态库库的优先级提升一下，目前我们公司已经尽量不用mdk了，目前使用是LLVM-ET-Arm 或者 arm-none-eabi-gcc