dji-sdk/Payload-SDK #252: Custom payload - exposure settings

GregCornis, 2025-03-12:
We're developing a payload with a custom camera (Canon) and we'd like to be able to change its exposure settings from DJI Pilot : ISO / Aperture / Shutter speed. We've followed the documentation to register a `commonHandler`, which works well to take pictures and receive the video stream.
However, there doesn't seem to be any way to declare that a custom camera payload can handle manual exposure settings. Is there a way to do that ?

[image]
_We would like to have these settings for a custom gimbal camera_

tuyaliang, 2025-03-13:
> We're developing a payload with a custom camera (Canon) and we'd like to be able to change its exposure settings from DJI Pilot : ISO / Aperture / Shutter speed. We've followed the documentation to [...]

You can use custom widget to achieve it.

tuyaliang, 2025-03-13:
[image]

GregCornis, 2025-03-13:
Ok thanks, so there is no way to have a more "native" setting ? So that we can use [KeyShutterSpeed](https://developer.dji.com/api-reference-v5/android-api/Components/IKeyManager/Key_Camera_CameraKey.html#key_camera_shutterspeed_inline) from the MSDK for example ?

tuyaliang, 2025-03-14:
> Ok thanks, so there is no way to have a more "native" setting ? So that we can use [KeyShutterSpeed](https://developer.dji.com/api-reference-v5/android-api/Components/IKeyManager/Key_Camera_CameraKey.html#key_camera_shutterspeed_inline) from the MSDK for example ?

Yes, not at the moment

tuyaliang, 2025-03-14:
To meet the developer's control requirements for camera-like payloads, PSDK provides interfaces for controlling the camera to perform functions such as taking pictures, recordings, zooming, and focusing. Light and other functions, and then by registering the PSDK camera interface, a fully functional camera payload device is developed. By using DJI Pilot and the Mobile App developed based on MSDK, users can control the camera payload device developed with PSDK to execute the specified Action to obtain information and resources in the payload device.

Basic functions: set camera mode, take pictures, video, get camera status, SD card management
Advanced functions: zoom, metering, focus, video streaming, playback download, media library management
[#](https://developer.dji.com/doc/payload-sdk-tutorial/cn/#introduction-to-basic-concepts)