dji-sdk/Payload-SDK #254: Error when runing Flight Control samples

Maxuos4, 2025-04-02:
Hello, i am experimenting with the payload sdk sample in a 350 RTK runing the payload on a raspberry pi 5 with raspberry pi OS and while the fc Subscription sample runs fine whenever i try the flight control sample I get this error messages: 

Trying takeoff-landing sample:
[44.252][errno]-[Error]-[DjiError_GetErrorMsgElements:85) Search error object failed. 
[44.252][errno]-[Error]-[DjiError_GetErrorMsgElements:102) Print error message failed, error code is 0x1D00000033. 
[44.252][flight]-[Error]-[DjiFlightController_Action_M350:384) (null)(null)(null) 
[44.252][user]-[Error]-[DjiTest_FlightControlMonitoredTakeoff:1244) Request to take off failed, error code: 0x00000033 
[44.252][user]-[Error]-[DjiTest_FlightControlTakeOffLandingSample:307) Take off failed 

Arrest Flying Sample:

[28.495][errno]-[Error]-[DjiError_GetErrorMsgElements:85) Search error object failed. 
[28.495][errno]-[Error]-[DjiError_GetErrorMsgElements:102) Print error message failed, error code is 0x1D00000033. 
[28.495][flight]-[Error]-[DjiFlightController_Action_M350:384) (null)(null)(null) 

I have been looking but i can´t find where they come from my onlly clue is that the problem is shown when calling the DjiFlightController_TurnOnMotors() and DjiTest_FlightControlMonitoredTakeoff() functions.

Thanks for any kind off help

dji-dev, 2025-04-03:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138275'>Zendesk ticket #138275</a>:<br><br>Hello, have you used the flight control example in the C++ sample? This seems to be caused by the failure to obtain control.
<br><p><em><small>°°°</small></em></p>

Maxuos4, 2025-04-03:
Yes i used de c++ executable sample and the log output for the DjiFlightController_ObtainJoystickCtrlAuthority() function is just the green step 1 text with no error message

dji-dev, 2025-04-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138275'>Zendesk ticket #138275</a>:<br><br>Hello, thank you for your supplement. What is the PSDK version number you use? Can you record a short video? Let us take a look at your test environment and specific situation.
<br><p><em><small>°°°</small></em></p>

Maxuos4, 2025-04-08:
Hello, I am using the master version of PSDK from github with the latest update, here are images of the error message plus my setup:

[image]
[image]
[image]
[image]

dji-dev, 2025-04-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138275'>Zendesk ticket #138275</a>:<br><br>Hello, have you opened the simulator software in DJI Assistant2 software on your computer?
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-04-10:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138275'>Zendesk ticket #138275</a>:<br><br>Hello, we have confirmed that this error code is caused by the aircraft not undergoing real-name authentication. You can confirm whether real-name authentication has been performed on the pilot. If you are not sure how to authenticate, please contact support@dji.com and we will have someone to provide guidance for you.
<br><p><em><small>°°°</small></em></p>