dji-sdk/Payload-SDK #255: Cannot initialize SDK due to missing SDK Certified Chip in Matrice 4E

BackLogers, 2025-04-05:
I’m currently testing the official PSDK example for STM32 from your SDK repository, running on a DJI Matrice 4E drone. The SDK initializes communication properly, detects the aircraft type and mount position, but ultimately fails during the authentication phase due to the lack of an SDK Certified Chip (CC).

Below is the log output:
[32m[2.454][core]-[Info]-[DjiCore_Init:106) Payload SDK Version : V3.10.0-beta.0-build.2190 Jan  9 2025 21:49:56 [0m
[32m[4.875][adapter]-[Info]-[DjiAccessAdapter_Init:313) Identify aircraft series is Mavic 4 Series [0m
[32m[4.875][adapter]-[Info]-[DjiAccessAdapter_Init:352) Identify mount position type is Extension-Lite Port Type [0m
[32m[4.876][adapter]-[Info]-[DjiAccessAdapter_Init:475) Identity uart0 baudrate is 921600 bps [0m
[32m[4.877][adapter]-[Info]-[DjiPayloadNegotiate_Init:134) No need negotiate device info [0m
[32m[4.879][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:493) Updating dji sdk policy file... [0m
[33m[4.933][linker]-[Warn]-[DjiCommand_DealCmd:845) Received invalid ack,<0x01> 0x0300->0x0A07 0x03FD [0m
[32m[5.879][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:501) Update dji sdk policy file successfully [0m
[32m[5.905][infor]-[Info]-[DjiAircraftInfo_GetAircraftTypeByCameraVersion:653) get aircraft type error! cmd return code: 224 [0m
[32m[6.906][infor]-[Info]-[DjiAircraftInfo_Init:277) Try to get aircraft type again... [0m
[32m[6.948][core]-[Info]-[DjiCore_Init:166) Identify AircraftType = Mavic 4E, MountPosition = Extension-Lite Port, SdkAdapterType = None [0m
[31m[7.318][auth]-[Error]-[DjiAuthAdapter_Init:80) Connect DJI SDK CC device failed, errno: 0x30000002. [0m
[31m[7.318][auth]-[Error]-[DjiAuth_SdkEncryptionChipInit:74) DJI SDK CC init failed, errno = 0x30313000 [0m
[31m[7.319][auth]-[Error]-[DjiAuth_Sha256RsaVerify:109) SDK encryption chip init failed, errno = 0x200036F4 [0m
[31m[7.319][core]-[Error]-[DjiCore_Init:221) Auth userinfo failed, stat:236 [0m
[31m[7.319][user]-[Error]-[DjiUser_StartTask:181) core init error [0m

My questions:
- Is there any way to enable CC simulation mode for testing the STM32 PSDK example on Matrice 4E without the physical encryption chip?
- If not, where can I obtain a compatible SDK Certified Chip? Is it possible to purchase this module directly from DJI or an authorized supplier?

Once I have the SDK CC chip:
- Will it be automatically detected by the example code if connected via I2C (assuming correct wiring)?
- Or do I need to modify specific source files to integrate it (e.g. via custom init functions)?

dji-dev, 2025-04-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138544'>Zendesk ticket #138544</a>:<br><br>Hello, I'm sorry, there is no simulator available for debugging CC certified chips at present. Only physical chip debugging is supported. You can contact your local agent to purchase. As long as you connect the CC chip, PSDK will automatically find and detect it. You can refer to the debugging method here: https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/sdk-cc.html
<br><p><em><small>°°°</small></em></p>

BackLogers, 2025-04-13:
Thanks for your help. I managed to order the CC chip from a local agent. 

Unfortunately I have another problem. The documentation does not contain the exact dimensions of the hardware packaging of this chip. I cannot find the documentation for the mentioned chip packaging "05D4 DFN8 2x3" Can you provide the exact footprint for this chip?

dji-dev, 2025-04-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138544'>Zendesk ticket #138544</a>:<br><br>Hello, would you please provide an email address? We will send the information directly to your email address.
<br><p><em><small>°°°</small></em></p>

BackLogers, 2025-04-15:
Sure, here's my email address:
b_srebro@atar.edu.pl

dji-dev, 2025-04-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/138544'>Zendesk ticket #138544</a>:<br><br>Hello, we have sent it, please check it. It is sent to you by the user of dev@dji.com.
<br><p><em><small>°°°</small></em></p>

closed 2025-04-22 by BackLogers

leonid-septier, 2025-06-30:
Can you please also provide the footprint to me, too ? leonid@septier.com