dji-sdk/Payload-SDK #260: H264 stream corruption after camera switch using DjiLiveView_StartH264Stream

AndreaSanchini, 2025-04-23:
Hi, 
I am working with the PSDK on a DJI M300 RTK with H20 payload, using psdk version 3.8.1. I am using the wide camera to help the drone navigate, but I occasionally need to switch to the zoom camera  (see #241)  to set some parameters like the focus value.

Currently, I switch the active camera  to zoom using `DjiLiveView_StartH264Stream`, set the desired parameters (e.g. using `DjiCameraManager_GetFocusRingValue`), then switch back the active camera to wide with the same function. It's worth mentioning that in my code I repeatedly call `DjiLiveview_RequestIntraframeFrameData` to help my hardware decoder decode the DJIH264 stream. 

The issue is that sometimes this process causes the live stream to corrupt , which is visible on the remote controller. Given this issue I'd like to ask the following questions:

- Could the corruption be caused by an incorrect usage of `DjiLiveview_RequestIntraframeFrameData`?
- Should I explicitly stop the current stream with `DjiLiveview_StopH264Stream` before starting another one, or is switching active camera via `DjiLiveView_StartH264Stream` sufficient?
- [Documentation for release 3.11.1](https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/advanced-function/liveview.html#i-frame-sometimes-appears-in-h20-h20t-h-264-stream-in-gdr-encoding-format) says an I-frame is inserted when switching camera from remote controller, does the same happen when starting a stream with `DjiLiveView_StartH264Stream` or not?

Thanks,
Andrea

dji-dev, 2025-04-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/140827'>Zendesk ticket #140827</a>:<br><br>Hello, first of all, before you switch to another video source, you need to stop the current video stream transmission, and then start the new stream. This is the correct process. Secondly, when calling the PSDK API interface to start the stream, an I frame will be actively inserted.
<br><p><em><small>°°°</small></em></p>

AndreaSanchini, 2025-05-05:
I found that the PSDK provides the `DjiCameraManager_SetStreamSource` function. Is this equivalent to stopping the stream for the current camera and starting it for another one using `DjiLiveview_StopH264Stream` and `DjiLiveview_StartH264Stream`?

dji-dev, 2025-05-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/140827'>Zendesk ticket #140827</a>:<br><br>Hello, the DjiCameraManager_SetStreamSource function allows you to switch the lens source. The usage method is consistent with that mentioned in this article.
https://sdk-forum.dji.net/hc/zh-cn/articles/25550984354969-M3E-M3T%E5%A6%82%E4%BD%95%E7%94%A8PSDK%E8%8E%B7%E5%8F%96%E5%8F%98%E7%84%A6%E8%A7%86%E9%A2%91%E6%B5%81
<br><p><em><small>°°°</small></em></p>

AndreaSanchini, 2025-05-06:
Hello, thank you for your reply. 

I read the article, but I'm not completely sure whether `DjiCameraManager_SetStreamSource` is appropriate for my use case. I normally start an H264 stream from the wide camera on an H20 payload, but occasionally need to switch to the zoom camera to adjust parameters.

Is it necessary to stop the stream and restart it with the new source, or can I simply call `DjiCameraManager_SetStreamSource` while the stream is active?
I tried both approaches and they both work, but I’d like to understand the correct or recommended procedure for this scenario, if there is one.

Thank you

dji-dev, 2025-05-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/140827'>Zendesk ticket #140827</a>:<br><br>Hi, sorry, maybe I didn't explain it clearly, you can follow this flow, I will show you how to switch from wide angle to zoom stream;

DjiLiveview_StartH264Stream((E_DjiLiveViewCameraPosition) mountPosition,
DJI_LIVEVIEW_CAMERA_SOURCE_H20_WIDE,
DjiTest_PayloadCameraStreamCallback);

wait …..

DjiLiveview_StopH264Stream((E_DjiLiveViewCameraPosition) mountPosition,
DJI_LIVEVIEW_CAMERA_SOURCE_H20_WIDE,
DjiTest_PayloadCameraStreamCallback);

wait …..

DjiLiveview_StartH264Stream((E_DjiLiveViewCameraPosition) mountPosition,
DJI_LIVEVIEW_CAMERA_SOURCE_H20_ZOOM,
DjiTest_PayloadCameraStreamCallback);

wait…..

DjiLiveview_StopH264Stream((E_DjiLiveViewCameraPosition) mountPosition,
DJI_LIVEVIEW_CAMERA_SOURCE_H20_ZOOM,
DjiTest_PayloadCameraStreamCallback);
<br><p><em><small>°°°</small></em></p>

AndreaSanchini, 2025-05-07:
Hi, thanks again for your reply.
When you say I need to wait between start/stop calls, do you mean I can proceed as soon as `T_DjiReturnCode == DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS`, or should I also wait for a fixed amount of time?

dji-dev, 2025-05-08:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/140827'>Zendesk ticket #140827</a>:<br><br>You can execute as quickly as possible, and you don't need to add waiting logic.
<br><p><em><small>°°°</small></em></p>