dji-sdk/Payload-SDK #262: multiple definition of `SystemClock_Config'

dzx815, 2025-05-09:
3.11.1
payload-sdk-arm-none-eabi-gcc

编译报错
/home/leon/workspace/payload-sdk/modules/authentication/ciu98/huada_lib/src/util.c:140: multiple definition of `SystemClock_Config';

dji-dev, 2025-05-09:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142136'>Zendesk ticket #142136</a>:<br><br>很抱歉为您带来不便，麻烦您从此处下载修改过后的静态库：
https://pan-sec.djicorp.com/s/KG4xF2T2SjZKmFZ
<br><p><em><small>°°°</small></em></p>

dzx815, 2025-05-13:
确认已解决

dji-dev, 2025-05-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142136'>Zendesk ticket #142136</a>:<br><br>您好，抱歉，这次确定给您的是3.11.1的https://pan-sec.djicorp.com/s/KG4xF2T2SjZKmFZ
<br><p><em><small>°°°</small></em></p>

dzx815, 2025-05-16:
通过日志打印出来，还是3.11，前天是有截图提交的，可能网络原因没有显示出来，这里再提交一次。

dji-dev, 2025-05-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142136'>Zendesk ticket #142136</a>:<br><br>您好，我们与内部团队进行了确认，很抱歉，应该是服务器配置的问题，您再尝试此处：
https://pan-sec.djicorp.com/s/pqpEdofAFroqA4w
<br><p><em><small>°°°</small></em></p>

dzx815, 2025-05-16:
[image]

这不就回到了最开始的问题，又是重复定义/home/leon/workspace/payload-sdk/modules/authentication/ciu98/huada_lib/src/util.c:140: multiple definition of `SystemClock_Config';

dji-dev, 2025-05-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142136'>Zendesk ticket #142136</a>:<br><br>很抱歉为您带来不好的体验，麻烦您使用此库：https://pan-sec.djicorp.com/s/GQgeccYJY5C9sN3
另外版本您看看是否为3.11.1
<br><p><em><small>°°°</small></em></p>

dzx815, 2025-05-19:
[image]

确认了版本是3.11.1

closed 2025-05-19 by dzx815

dji-dev, 2025-05-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142136'>Zendesk ticket #142136</a>:<br><br>您好，感谢您的反馈，麻烦您这边再尝试下看是否还有其他的问题。
<br><p><em><small>°°°</small></em></p>