dji-sdk/Payload-SDK #263: Successful setup of UART and bulk but I can't get the frame when i run the sample

Y1annis, 2025-05-13:
Hello! I attached a screenshot of the terminal. Currently no camera is connected but I tried to get a frame from the FPV but the frame does not appear. Do you know what is the issue?

[image]

dji-dev, 2025-05-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142554'>Zendesk ticket #142554</a>:<br><br>Hello, we have a knowledge base dedicated to this, you may refer to it here: https://sdk-forum.dji.net/hc/zh-cn/articles/43810493122457-PSDK-C-sample-FPV%E6%97%A0%E6%B3%95%E6%98%BE%E7%A4%BA%E8%A7%86%E9%A2%91%E6%B5%81
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-05-16:
Hello, I also wanted to ask which baud_rate is recommended, because I set it to 921600 and sometimes the UART is not communicating with the drone and I have to replug the USB. Thank you!

dji-dev, 2025-05-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142554'>Zendesk ticket #142554</a>:<br><br>Hello, we do not have a recommended baud rate or a specified baud rate. This is determined by the situation of your board and the number of enabled functions. For example, if you subscribe to many TOPICs in PSDK, we would recommend that you use a baud rate of 460800 or above. If your board does not support such a high baud rate, it is generally around 230400.
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-05-19:
Okay thank you! One more question, which sample is recommended? sample_c or sample_c++

dji-dev, 2025-05-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142554'>Zendesk ticket #142554</a>:<br><br>Hello, depending on your habits, we think C sample is easier to use.
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-05-19:
Hello again! And both of those samples have the same functionalities?

dji-dev, 2025-05-19:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142554'>Zendesk ticket #142554</a>:<br><br>Hello, actually the available functions are the same, but the C++ sample is more comprehensive in demonstration.
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-05-19:
I tried what is showed in the link you sent me: https://sdk-forum.dji.net/hc/zh-cn/articles/43810493122457-PSDK-C-sample-FPV%E6%97%A0%E6%B3%95%E6%98%BE%E7%A4%BA%E8%A7%86%E9%A2%91%E6%B5%81
But unfortunately I still cannot receive stream from the camera. It says that the stream is saved in a file but it seems that nothing is saved.