dji-sdk/Payload-SDK #264: Inquiry About Anti-Collision Light Control via DJI PSDK for M30,M300,M350,and M4

khapaladin, 2025-05-15:
Hello,
The FAA has passed a regulation requiring that strobe lights on UAVs be turned on via the PSDK. This is critical for us because we frequently operate our drones beyond the range of the remote controller (RC), and most of our pilots control flights through our own control interface instead of the RC. As a result, we need the ability to programmatically control all strobe lights—specifically turning them on and off—via the PSDK.
Is this functionality supported? This is an urgent requirement for our operations.

dji-dev, 2025-05-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142890'>Zendesk ticket #142890</a>:<br><br>Hello, we understand that PSDK has no adaptation plan for this in the short term. What models do you need to control? Are they all drones in all industry series?
<br><p><em><small>°°°</small></em></p>

khapaladin, 2025-05-16:
Hi!
We need to control all of the matrice series that support the PSDK. M30, M300, M350, M4s, and so on.

dji-dev, 2025-05-16:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/142890'>Zendesk ticket #142890</a>:<br><br>Hello, OK, we noticed that you sent emails simultaneously. To better track this issue and avoid confusion in communication, we recommend that you communicate in the same ticket.
<br><p><em><small>°°°</small></em></p>