dji-sdk/Payload-SDK #268: Jetson Xavier NX can't connect after i terminate the program

Y1annis, 2025-05-27:
Hello! I noticed that when i press ctrl + c when I successfully execute psdk the next time i try to connect i cant get control over the drone and I get this error: 
[image]

In order to reconnect I have to remove and plug in the usb again, sometimes I also get the UART error and I have to replug the USB.

Thank you!

dji-dev, 2025-05-27:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/144195'>Zendesk ticket #144195</a>:<br><br>Hello, which version of PSDK are you using? And which static library are you using? We are unable to reproduce your problem locally.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-05-28:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/144195'>Zendesk ticket #144195</a>:<br><br>Hello, you can see many names of compilation chain tools in the directory Payload-SDK/psdk_lib/lib of the code. You can see which library you are using when cmake, which is convenient for us to locate your problem. In addition, your PSDK version is not the latest. In order to avoid the impact of known issues, we recommend that you update to the latest version. At the same time, the drone firmware needs to be kept up to date.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-06-03:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/144195'>Zendesk ticket #144195</a>:<br><br>Hello, thank you for your feedback. We are glad that your problem has been solved. If you have any other questions, you are welcome to contact us again.
<br><p><em><small>°°°</small></em></p>