dji-sdk/Payload-SDK #269: Port Application from M300 to M350 - Problems with USB Bulk Connection on Lattepanda Delta 3 (x86)

mseidler79, 2025-06-04:
Hello,
we sucessfully running a application on a Matrix M300. We want to port the application to the M350. Hardware is a Lattepanda Delta 3 which is an Intel Celeron N5150 board running Linux. As connection to the drone we are using the office DJI developement kit.

UART connection to the M350 is working fine. But we are struggeling to get the USB Bulk connection running. When using the ifconfig command we see the Ethernet interface running via the USB config and ping the drone on the IP address 192.168.44.2 is sucsessfull.

Could you support us please to get the USB bulk connection between M350 and Lattapanda Delta 3 running. How to add the USB bulk interfaces to the Lattepanda? At the moment the M350 is connected via the DJI developement kit USB C port to the Lattepanda USB C port. Thank you for your support!

Output ifconfig:

enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.26.251.100  netmask 255.255.255.0  broadcast 172.26.251.255
        inet6 fe80::b99e:3a8a:5ca6:3fd2  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:4c:05:f8:39  txqueuelen 1000  (Ethernet)
        RX packets 3829  bytes 988802 (988.8 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 987  bytes 176498 (176.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0x7fd00000-7fdfffff

enx00e100001835: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 00:e1:00:00:18:35  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

enxee8ba72a7119: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.44.3  netmask 255.255.255.0  broadcast 192.168.44.255
        inet6 fe80::bc07:137a:97c9:5932  prefixlen 64  scopeid 0x20<link>
        ether ee:8b:a7:2a:71:19  txqueuelen 1000  (Ethernet)
        RX packets 16  bytes 1518 (1.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 55  bytes 9544 (9.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Lokale Schleife)
        RX packets 945  bytes 72898 (72.8 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 945  bytes 72898 (72.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Output lsusb:
Bus 002 Device 003: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 002 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 001 Device 010: ID 2ca3:001f DJI Technology Co., Ltd. e1e
Bus 001 Device 008: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 006: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 004: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 007: ID 046a:00b6 Cherry GmbH USB Mouse
Bus 001 Device 005: ID 046a:0023 Cherry GmbH Keyboard
Bus 001 Device 003: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Output communiction drone to application:

[2.002][core]-[Info]-[DjiCore_Init:106) Payload SDK Version : V3.9.1-beta.0-build.2090
[2.136][adapter]-[Info]-[DjiAccessAdapter_Init:231) Identify aircraft series is Matrice 350 Series
[2.136][adapter]-[Info]-[DjiAccessAdapter_Init:264) Identify mount position type is Extension Port Type
[2.139][adapter]-[Info]-[DjiAccessAdapter_Init:365) Identity uart0 baudrate is 921600 bps
[3.144][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8a9d>Command async send retry: index = 0, retryTimes = 3, 0x0A06->0x0701 0x0087
[6.047][adapter]-[Info]-[DjiPayloadNegotiate_Init:228) Waiting payload negotiate finish.
[8.745][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:474) Updating dji sdk policy file...
[9.745][core]-[Info]-[DjiIdentityVerify_UpdatePolicy:482) Update dji sdk policy file successfully
[9.763][core]-[Info]-[DjiCore_Init:174) Identify AircraftType = Matrice 350 RTK, MountPosition = Extension Port, SdkAdapterType = None
[9.804][core]-[Info]-[DjiCore_ApplicationStart:238) Start dji sdk application
[9.804][user]-[Info]-[applicationStart:909) Application start.
[11.804][flight]-[Info]-[DjiFlightController_RegisterLinkerObj_M350:139) Init M350 flight controller linker successfully.
[11.903][channel]-[Error]-[DjiStreamChannelUsbBulk_AddPort:473) Init usb bulk channel failed, ret:000000EC.
[11.903][channel]-[Error]-[DjiStreamChannelUsbBulk_Create:132) Add usb bulk port failed, ret:000000EC.
[11.903][downloader]-[Error]-[DjiCameraManager_MediaDownloaderInit:423) Init stream channel failed, ret:236.
[11.903][cmu]-[Error]-[DjiCameraManager_Init:252) Init media downloader failed, error code: 0x000000E3.
[11.903][user]-[Error]-[Psdk:350) Init camera manager failed, error code: 0x000000E3
[12.404][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8af8>Command async send retry: index = 0, retryTimes = 3, 0x0A06->0x0100 0x0001
[12.905][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8af8>Command async send retry: index = 0, retryTimes = 2, 0x0A06->0x0100 0x0001
[13.406][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8af8>Command async send retry: index = 0, retryTimes = 1, 0x0A06->0x0100 0x0001
[13.907][linker]-[Error]-[DjiCommand_SendAsyncHandle:975) Command async send error 0
[13.907][linker]-[Error]-[DjiCommand_SendSync:550) sendSync callback timeout
[13.907][cmu]-[Error]-[DjiCameraManager_GetCameraType:456) Get mount position 1 camera version error,error code: 0x000000E1.
[13.907][cmu]-[Error]-[DjiCameraManager_GetCameraType:457) Execution timeout. Suggestion : Please contact DJI for help.
[13.907][user]-[Error]-[Psdk:365) Get mounted position 21939 camera's type failed, error code: 0x000000E1

[14.408][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b29>Command async send retry: index = 0, retryTimes = 3, 0x0A06->0x0102 0x0001
[14.909][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b29>Command async send retry: index = 0, retryTimes = 2, 0x0A06->0x0102 0x0001
[15.410][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b29>Command async send retry: index = 0, retryTimes = 1, 0x0A06->0x0102 0x0001
[15.911][linker]-[Error]-[DjiCommand_SendAsyncHandle:975) Command async send error 0
[15.911][linker]-[Error]-[DjiCommand_SendSync:550) sendSync callback timeout
[15.911][cmu]-[Error]-[DjiCameraManager_GetCameraType:456) Get mount position 2 camera version error,error code: 0x000000E1.
[15.911][cmu]-[Error]-[DjiCameraManager_GetCameraType:457) Execution timeout. Suggestion : Please contact DJI for help.
[15.911][user]-[Error]-[Psdk:365) Get mounted position 21939 camera's type failed, error code: 0x000000E1

[16.412][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b58>Command async send retry: index = 0, retryTimes = 3, 0x0A06->0x0104 0x0001
[16.913][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b58>Command async send retry: index = 0, retryTimes = 2, 0x0A06->0x0104 0x0001
[17.414][linker]-[Warn]-[DjiCommand_SendAsyncHandle:963) <0x8b58>Command async send retry: index = 0, retryTimes = 1, 0x0A06->0x0104 0x0001
[17.915][linker]-[Error]-[DjiCommand_SendAsyncHandle:975) Command async send error 0
PSDK: Cam1: Unknown / Cam2: Unknown / Cam3: Unknown
[17.915][linker]-[Error]-[DjiCommand_SendSync:550) sendSync callback timeout
[17.915][cmu]-[Error]-[DjiCameraManager_GetCameraType:456) Get mount position 3 camera version error,error code: 0x000000E1.
[17.915][cmu]-[Error]-[DjiCameraManager_GetCameraType:457) Execution timeout. Suggestion : Please contact DJI for help.
[17.915][user]-[Error]-[Psdk:365) Get mounted position 21939 camera's type failed, error code: 0x000000E1

Camera Gimbal: Status Camera / Gimbal: 1
[18.416][channel]-[Error]-[DjiStreamChannelUsbBulk_AddPort:473) Init usb bulk channel failed, ret:000000EC.
[18.416][channel]-[Error]-[DjiStreamChannelUsbBulk_Create:132) Add usb bulk port failed, ret:000000EC.
[18.416][liveview]-[Error]-[DjiLiveview_BulkRegisterCallback:1021) Init usb bulk stream channel failed, ret:236.
[18.416][liveview]-[Error]-[DjiLiveview_Init:296) Init fpv usb bulk stream channel failed, ret:227.
[18.416][user]-[Error]-[videoStartH264Stream:2882) Liveview init failed, error code: 0x000000EC
[18.924][core]-[Error]-[DjiLiveview_FindLiveviewSource:1127) Can not found the liveview source. Probably the payload is not mounted on position 7 correctly. Please check the mount position of payload.
[18.924][liveview]-[Error]-[DjiLiveview_StartH264Stream:546) Can't find source, errno: 0x00000100.

mseidler79, 2025-06-10:
Sorry have just seen that the code of the script will be intrepted as BB Tags. So I put it into code tags.

mseidler79, 2025-06-10:
Just found out that I have to load the kernel modul (modprobe libcomposite) to get the UDC running. Now I'm struggling where to finde the startup_usb_bulk .c file which is mentioned in the script in the folder kyle/dji/work. Could you please give me the link to download the files. Thanks!

dji-dev, 2025-06-11:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/144977'>Zendesk ticket #144977</a>:<br><br>Hello, no, this is just our default configuration. This file is generated by compiling startup_usb_bulk.c by yourself. You can find the startup_usb_bulk.c file in this file. https://terra-1-g.djicdn.com/71a7d383e71a4fb8887a310eb746b47f/psdk/e-port/usb-bulk-configuration-reference.zip
<br><p><em><small>°°°</small></em></p>

mseidler79, 2025-06-12:
Hello Leon,
usb_bulk_start.sh and startup_bulk.c are done. Service on Linux for autostart the usb_bulk_start.sh is set and it is automatically starting on boot.

The usb_bulk_start.sh script leads at the end to the problem that there is no device USB0. This is also not seen in LSUSB or IFCONFIG.

How to get the device USB0?

Fault messages usb_bulk_start.sh:
`interface usb0 does not exist!
usb0: ERROR while getting interface flags: no such device
usb0: ERROR while getting interface flags: no such device`

These messages are created by the commands:
`/sbin/brctl addif lp4br0 usb0
/sbin/ifconfig usb0 down
/sbin/ifconfig usb0 up`

lsusb:
`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0026 Intel Corp. AX201 Bluetooth
Bus 001 Device 005: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 004: ID 2341:8036 Arduino SA Leonardo (CDC ACM, HID)
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

ifconfig:
`enp1s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.26.251.100  netmask 255.255.255.0  broadcast 172.26.251.255
        inet6 fe80::b99e:3a8a:5ca6:3fd2  prefixlen 64  scopeid 0x20<link>
        ether 00:e0:4c:05:f8:39  txqueuelen 1000  (Ethernet)
        RX packets 7931  bytes 1502843 (1.5 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3053  bytes 602318 (602.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0x7fd00000-7fdfffff

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Lokale Schleife)
        RX packets 3341  bytes 189634 (189.6 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3341  bytes 189634 (189.6 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lp4br0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.55.1  netmask 255.255.255.0  broadcast 192.168.55.255
        ether 06:b1:c0:77:f6:0d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0`

usb_bulk_start.sh script:
`#!/bin/bash

# set -x

sudo modprobe libcomposite
sleep 2

cd /sys/kernel/config/usb_gadget/
mkdir -p lattepanda

cd lattepanda
echo 0x0955 > idVendor # Linux Foundation
echo 0x7020 > idProduct # Multifunction Composite Gadget
echo 0x0001 > bcdDevice # v1.0.0

echo 0x0200 > bcdUSB # USB2

echo 0xEF > bDeviceClass
echo 0x02 > bDeviceSubClass
echo 0x01 > bDeviceProtocol

mkdir -p strings/0x409
echo "abcdefg1234567890" > strings/0x409/serialnumber
echo "lattepanda" > strings/0x409/manufacturer
echo "LP" > strings/0x409/product

cfg=configs/c.1
mkdir -p "${cfg}"
echo 0x80 > ${cfg}/bmAttributes
echo 250 > ${cfg}/MaxPower

cfg_str=""
udc_dev=fe980000.usb

# The IP address shared by all USB network interfaces created by this script.
net_ip=192.168.55.1
# The associated netmask.
net_mask=255.255.255.0

# Note: RNDIS must be the first function in the configuration, or Windows'
# RNDIS support will not operate correctly.
enable_rndis=1
if [ ${enable_rndis} -eq 1 ]; then
cfg_str="${cfg_str}+RNDIS"
func=functions/rndis.usb0
mkdir -p "${func}"
ln -sf "${func}" "${cfg}"

# Informs Windows that this device is compatible with the built-in RNDIS
# driver. This allows automatic driver installation without any need for
# a .inf file or manual driver selection.
echo 1 > os_desc/use
echo 0xcd > os_desc/b_vendor_code
echo MSFT100 > os_desc/qw_sign
echo RNDIS > "${func}/os_desc/interface.rndis/compatible_id"
echo 5162001 > "${func}/os_desc/interface.rndis/sub_compatible_id"
ln -sf "${cfg}" os_desc
fi

enable_bulk=1
if [ ${enable_bulk}-eq1 ]; then
mkdir -p /dev/usb-ffs

cfg_str="${cfg_str}+BULK1"
mkdir -p /dev/usb-ffs/bulk1
func=functions/ffs.bulk1
mkdir -p "${func}"
ln -sf "${func}" "${cfg}"
mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk1 /dev/usb-ffs/bulk1
/home/c5uav/airstation/usb-bulk-configuration-reference/startup_bulk /dev/usb-ffs/bulk1 &
sleep 3

cfg_str="${cfg_str}+BULK2"
mkdir -p /dev/usb-ffs/bulk2
func=functions/ffs.bulk2
mkdir -p "${func}"
ln -sf "${func}" "${cfg}"
mount -o mode=0777 -o uid=2000 -o gid=2000 -t functionfs bulk2 /dev/usb-ffs/bulk2
/home/c5uav/airstation/usb-bulk-configuration-reference/startup_bulk /dev/usb-ffs/bulk2 &
sleep 3
fi

mkdir -p "${cfg}/strings/0x409"
echo "${cfg_str:1}" > "${cfg}/strings/0x409/configuration"

udevadm settle -t 5 || :
ls /sys/class/udc > UDC

/sbin/brctl addbr lp4br0
/sbin/ifconfig lp4br0 ${net_ip} netmask ${net_mask} up

if [ ${enable_rndis} -eq 1 ]; then
/sbin/brctl addif lp4br0 usb0
/sbin/ifconfig usb0 down
/sbin/ifconfig usb0 up
fi

exit 0`

**Thank you for your support !**

dji-dev, 2025-06-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/144977'>Zendesk ticket #144977</a>:<br><br>Hello, it seems that your node is successfully pulled up. If you want to use the lsusb command, you need to first connect the development board as a Device to another Linux device. Of course, you can also connect the Device interface of the development board to the Host so that it can enumerate its own VID PID.
<br><p><em><small>°°°</small></em></p>

csPinKie, 2025-10-05:
> Agent comment from Leon in [Zendesk ticket #144977](https://djisdksupport.zendesk.com/agent/tickets/144977):Hello, it seems that your node is successfully pulled up. If you want to use the lsusb command, you need to first connect the development board as a Device to another Linux device. Of course, you can also connect the Device interface of the development board to the Host so that it can enumerate its own VID PID.
> 
> _°°°_

Hi, i am facing the same issue that i cant get the vid/pid from the dji e dev kit,  when having a connwction from the drone to the edev kit and then to the raspberry pi, lsusb only shows the uart adapter and 3 other usb controllers plus the usb hub.

But i dont get a vid/pid from the edev? How can i get this?