dji-sdk/Payload-SDK #271: Managed to receive camera stream on the DJI Matrice 300 but not on the Matrice 350

Y1annis, 2025-06-11:
Hello! I am currently using PSDK Version 3.11, Jetson Xavier NX, E-port development kit and the DJI Matrice 350. I succesfully configured uart and bulk  because I have the bulk1/ep1, bulk1/ep2 and bulk2/ep1, bulk2/ep2. When I run the liveview sample on the Matrice 300 I get the stream successfully. However, when I am using the Matrice 350 I can't get the stream (I dont get the pop with the frame). Do you know what causes this?

Thank you!

uzgit, 2025-06-11:
Hi Yiannis,

I am just a PSDK user, but I am subscribed to the PSDK GitHub repo, so I saw this issue. I made a setup script a while back to make the USB bulk functionality work on the E-port with a Matrice 350 and Raspberry Pi 4 and 5. You can check it out and see if it works for you. (Works on the Pi, might not work for Jetson series, but might give you an idea of how to set it up.)

File: https://github.com/uzgit/Payload-SDK/blob/master/configure.sh
From about lines lines 394 to 1333

There are more guidelines on the repo page: https://github.com/uzgit/Payload-SDK/

Y1annis, 2025-06-13:
Hello guys and than you for the answers! So I need to buy seperate hardware to run the uart+network device>

Y1annis, 2025-06-23:
@dji-dev Hello! Any input on this?

Y1annis, 2025-06-23:
Hello @dji-dev , isn't it weird that I can get frames from Matrice 300 but can't on the Matrice 350? On the Matrice 350 it is stuck on the press q to exit stream but no frame is shown. For the network card what is the setup? It is connected on the e-port and then ethernet is connected to the Jetson? Thank you for the answers

AndreaSanchini, 2025-06-23:
Hi @Y1annis,
concerning the uart and network setup on a jetson platform, try to have a look here https://aerostack2.github.io/_03_aerial_platforms/_dji_psdk/index.html.

dji-dev, 2025-06-24:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/145797'>Zendesk ticket #145797</a>:<br><br>Hello, this is the design. First of all, the links of M300 and M350 are different. M300 is the drone as a device connected to your development board (it does not require you to configure the BULK environment), and M350 is the drone as a host, and your development board is connected to the aircraft as a device, so you need to configure the BULK environment, otherwise the link will be unavailable.
<br><p><em><small>°°°</small></em></p>

Y1annis, 2025-06-25:
Thank you for the answer, it turns out that the jetson xavier nx should have been connected to the micro-usb port because it is the only port that can support device mode. Also i used the uart+network configuration through the jetson's micro usb

dji-dev, 2025-06-26:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/145797'>Zendesk ticket #145797</a>:<br><br>Hello, thank you for your supplement. May I ask if you have solved this problem? In fact, you can choose either method. You don't need to connect two links at the same time.
<br><p><em><small>°°°</small></em></p>