dji-sdk/Payload-SDK #274: UART connection failed on M400

akgongon, 2025-06-26:
I connected my payload developed with PSDK 3.7 to the Matrice 400, but a UART connection error occurs and the program terminates.
However, with the latest PSDK 3.11.1, it seems to be working fine.
Which PSDK version does the Matrice 400 support?

I have no choice but to use PSDK 3.7 because the latest PSDK has a bug when using CUSTOM-H264, as listed in the known issues.
When is this bug expected to be resolved?

Janustin, 2025-06-27:
> I connected my payload developed with PSDK 3.7 to the Matrice 400, but a UART connection error occurs and the program terminates. However, with the latest PSDK 3.11.1, it seems to be working fine. [...]

Hello friend，According to internal testing, the PSDK version that supports M400 is PSDK3.12.0 or PSDK3.13.0, which has not yet been officially released.

akgongon, 2025-06-27:
Can you provide an update on the CUSTOM-H264 function?
Is it going to be fixed, or is support no longer available?

akgongon, 2025-06-27:
Im sorry, and Thank you for your kindness, Janustin!
For now, I think I willl wait for the official release or response.

Janustin, 2025-06-27:
> Im sorry, and Thank you for your kindness, Janustin! For now, I think I willl wait for the official release or response.

It doesn’t matter, my friend. DJI has just released PSDK3.12.0 for M400. You can try updating it.