dji-sdk/Payload-SDK #276: Changing flight modes using DJI PSDK (T/S/N/A)

Y1annis, 2025-07-01:
Hello, is there a way to change the flight mode using DJI PDSK? I want to add a way for my drone to enter ATTI mode using my PSDK application? Is that possible?

Thank you!

Y1annis, 2025-07-10:
Is there any way to enable atti mode without the RC ? For example, can we block or disable GPS using the PSDK ? For spoofing or jamming attacks

dji-dev, 2025-07-11:
Agent comment from LIPING.ZHOU in <a href='https://djisdksupport.zendesk.com/agent/tickets/147964'>Zendesk ticket #147964</a>:<br><br>Dear Developer,

Hello and thank you for reaching out to DJI Innovations.

PSDK can only gain control in P mode. In a no-GPS environment, you need to set the drone to P mode and then use attitude angles and throttle control.

Here is the related documentation:
1.https://developer.dji.com/doc/payload-sdk-tutorial/en/function-set/advanced-function/flight-control.html
2.5.7 joystick飞行控制（flight_control）（https://sdk-forum.dji.net/hc/zh-cn/articles/5497460532889）

We hope that our solution meets your needs satisfactorily. We appreciate your email and wish you a wonderful day!

Best Regards,
DJI Innovations SDK Technical Support Team
<br><p><em><small>°°°</small></em></p>