dji-sdk/Payload-SDK #277: M400 liveview error

tuyaliang, 2025-07-01:
PSDK3.12 version, get live view error:

[image]

dji-dev, 2025-07-02:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/147972'>Zendesk ticket #147972</a>:<br><br>Hello, please refer to the question here. The third-party development board can only use basic functions. Only by using manifold3 development can you use advanced functions: https://developer.dji.com/doc/payload-sdk-tutorial/en/model-instruction/choose-develop-platform.html
<br><p><em><small>°°°</small></em></p>

tuyaliang, 2025-07-03:
Thanks,I have got fpv liveview and lidar together through Jetson xaiver nx via usb2.0.

[image]

[image]

whyscience, 2025-08-07:
Which connection do you use? DJI_USE_ONLY_NETWORK_DEVICE or DJI_USE_ONLY_USB_BULK_DEVICE?

```
#define DJI_USE_ONLY_UART                  (0)
#define DJI_USE_UART_AND_USB_BULK_DEVICE   (1)
#define DJI_USE_UART_AND_NETWORK_DEVICE    (2)
#define DJI_USE_ONLY_USB_BULK_DEVICE       (3)
#define DJI_USE_ONLY_NETWORK_DEVICE        (4)
```