dji-sdk/Payload-SDK #289: DJI PSDK setup to get high power from M400: experiencing tons of issues

HeJiahuanTD, 2025-08-05:
The current tutorial for setting up PSDK using E-port V2 development board (GD32F527ZMT7) on M400 is really lacking too much information. 
After downloading all required files, software, etc., including GD32 files, when compiling the bootloader, there is error on the __disable_fiq(); 
Then if going into compiling the project inside mdk_app, there are a list of errors:
[image]
I was able to get over these problems by deleting or adding some headers or commands, or changing some Target settings. Then at using J-link debugger, the tutorial uses the flash algorithm GD32F5xx 4608kB, while is something that simply doesn't exist in the newest GD32 pack.
I also managed to get over that by using provided flash algorithm and changing Read/Only memory area IROM1 start and size.
Eventually I managed to get both bootloader and the app projects compiled flashed to the board without error message, I was simply trying to change the voltage request to 24V (also editing some code in the test_power_management). Not working. Tried the debug mode in Keil, but it's automatically running and I could not get any useful log information or variable values.
Of course double checked my verification information, straight copy paste a few times, still not working. The Keil not showing any error when compiling or flashing, but the VCC port where I hope to get 24V only stays at 13.75V.

At this point I really have no idea where to move forward.

Jaminv1, 2025-09-21:
Hi, did you end up resolving this issue?I am also struggling with power supply from the M400.

DanielExp-ite, 2026-10-27:
Hi, in case it helps anyone with the same symptom (VCC stuck at ~13.7 V even though the 24 V request reports success)

Cause: the HPWR (high-power apply) pin was left unconnected. On our board it floated at ~3.2 V (logic high). The Linux sample's pin callback (`DjiTest_WriteHighPowerApplyPin`) is a no-op with the comment *"please pull up the HWPR pin state by hardware"*, which is misleading for the M400.

This is in DJI's official documentation: [How to apply high power output from M400](https://sdk-forum.dji.net/hc/en-us/articles/57637724146457-How-to-apply-high-power-output-from-M400) (needs a DJI developer forum login). The pin to pull down depends on the adapter (E-Port V2 coaxial cable, SkyPort V3 or E-Port V2 development board), so check the pin for your hardware there. See also the [Power Management tutorial](https://developer.dji.com/doc/payload-sdk-tutorial/en/basic-function/power-management.html).