dji-sdk/Payload-SDK #297: Intermittent grey screen on M400 when streaming H.264 (PSDK H264 DJI format) — stream drops after a while

LaVoute974, 2025-08-28:
I stream the Raspberry Pi desktop to a DJI aircraft via PSDK. On M350RTK the streamer works fine. On M400 (PSDK >= v3.12.0, using DJI_CAMERA_VIDEO_STREAM_TYPE_H264_DJI_FORMAT) the feed initially appears but after some time the controller screen goes grey and the stream stops showing video. No crash or obvious error printed by the streamer; the PSDK connection remains active.

Environment
- Raspberry Pi 5
- FFmpeg/libav: (5.1.6)
- PSDK: v3.12.0+ (tested)
- Target aircraft: DJI M400
- DJI Skyport V2 (V01.04.11.12)

Repro steps

1. On the Pi the streamer captures the desktop with:
   `ffmpeg -loglevel quiet -f x11grab -r <FPS> -s <WIDTH>x<HEIGHT> -i :0.0 -f rawvideo -pix_fmt yuv420p -`, FPS = 24, size of video : 640x480
2. Stream frames encoded with libx264 through FFmpeg encoder in the app and send frames via `DjiPayloadCamera_SendVideoStream()`.
3. Connect M400 controller and start viewing: video appears initially.
4. After ~tens of seconds, controller view becomes grey. Stream state queries continue to succeed; no fatal error in output.

Observed behavior
- Local dump plays with `ffplay` (desktop visible).
- On M400 controller: desktop initially visible, then grey after some time.
- No explicit error printed by the app; `DjiPayloadCamera_GetVideoStreamState()` returns values but the UI freezes/grey.

I modify the function `static T_DjiReturnCode SetFocusTarget(T_DjiCameraPointInScreen target)` to have the mouse with autofocus on the controller screen : 

```
static T_DjiReturnCode SetFocusTarget(T_DjiCameraPointInScreen target)
{
    USER_LOG_INFO("set focus target x:%.2f y:%.2f", target.focusX, target.focusY);
    memcpy(&s_cameraFocusTarget, &target, sizeof(T_DjiCameraPointInScreen));

    int mouse_x = 640 * target.focusX;
    int mouse_y = 480 * target.focusY;

    char command[100];
    sprintf(command, "xdotool mousemove %d %d click 1", mouse_x, mouse_y);

    system(command);

    return DJI_ERROR_SYSTEM_MODULE_CODE_SUCCESS;
}
```

And when the stream starts, I have the mouse, when it becomes grey and I'm no able to see the stream, the mouse still work so my program is running fine. 

I precise that on a M350, everything is fine, and the stream doesn't crash. 

If I stop the stream after the first crash and restart it, it may run for hours before crashing again, or it might only last a few minutes — the behavior is quite erratic.

@dji-dev maybe this is a drone firmware issue ?

dji-dev, 2025-08-28:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/154774'>Zendesk ticket #154774</a>:<br><br>Hello, what is the firmware version of your drone? We will further confirm this issue.
<br><p><em><small>°°°</small></em></p>

LaVoute974, 2025-08-28:
M400 version : V14.01.0906

LaVoute974, 2025-09-16:
Hello @dji-dev , any news ? Thanks in advance

LaVoute974, 2025-09-17:
Can I ask you what is your setup ? Do you use a raspberry pi ? With only a DJI SkyPort V2 ? Maybe I can give you my code for testing ?

dji-dev, 2025-09-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/154774'>Zendesk ticket #154774</a>:<br><br>Hello, we are using a Raspberry Pi 4B, and then connected to the drone via SkyportV2, selecting the link as UART&NETWORK, directly running the C sample test, and observing the load push screen on the remote control.
<br><p><em><small>°°°</small></em></p>

dji-dev, 2025-09-18:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/154774'>Zendesk ticket #154774</a>:<br><br>Hello, thank you for your supplement. If you have a necessary method, you can synchronize your code to us. Thank you again.
<br><p><em><small>°°°</small></em></p>