dji-sdk/Payload-SDK #302: Ubuntu 22.04 根据在文档-开发环境教程-make报错/usr/include/x86_64-linux-gnu/bits/floatn.h:86:9: error: ‘__float128’ does not name a type; did you mean ‘__cfloat128’?

mstsky115, 2025-09-02:
[ 44%] Building CXX object samples/sample_c++/platform/linux/manifold3/CMakeFiles/dji_sdk_demo_on_manifold3_cxx.dir/application/application.cpp.o
In file included from /usr/aarch64-linux-gnu/include/wchar.h:30,
                 from /usr/aarch64-linux-gnu/include/c++/11/cwchar:44,
                 from /usr/aarch64-linux-gnu/include/c++/11/bits/postypes.h:40,
                 from /usr/aarch64-linux-gnu/include/c++/11/iosfwd:40,
                 from /usr/aarch64-linux-gnu/include/c++/11/ios:38,
                 from /usr/aarch64-linux-gnu/include/c++/11/ostream:38,
                 from /usr/aarch64-linux-gnu/include/c++/11/iostream:39,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.hpp:31,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.cpp:26:
/usr/include/x86_64-linux-gnu/bits/floatn.h:86:9: error: ‘__float128’ does not name a type; did you mean ‘__cfloat128’?
   86 | typedef __float128 _Float128;
      |         ^~~~~~~~~~
      |         __cfloat128
In file included from /usr/aarch64-linux-gnu/include/c++/11/cwchar:44,
                 from /usr/aarch64-linux-gnu/include/c++/11/bits/postypes.h:40,
                 from /usr/aarch64-linux-gnu/include/c++/11/iosfwd:40,
                 from /usr/aarch64-linux-gnu/include/c++/11/ios:38,
                 from /usr/aarch64-linux-gnu/include/c++/11/ostream:38,
                 from /usr/aarch64-linux-gnu/include/c++/11/iostream:39,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.hpp:31,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.cpp:26:
/usr/aarch64-linux-gnu/include/wchar.h:407:8: error: ‘_Float128’ does not name a type; did you mean ‘_Float32x’?
  407 | extern _Float128 wcstof128 (const wchar_t *__restrict __nptr,
      |        ^~~~~~~~~
      |        _Float32x
/usr/aarch64-linux-gnu/include/wchar.h:524:8: error: ‘_Float128’ does not name a type; did you mean ‘_Float32x’?
  524 | extern _Float128 wcstof128_l (const wchar_t *__restrict __nptr,
      |        ^~~~~~~~~
      |        _Float32x
In file included from /usr/aarch64-linux-gnu/include/c++/11/cstdlib:75,
                 from /usr/aarch64-linux-gnu/include/c++/11/ext/string_conversions.h:41,
                 from /usr/aarch64-linux-gnu/include/c++/11/bits/basic_string.h:6608,
                 from /usr/aarch64-linux-gnu/include/c++/11/string:55,
                 from /usr/aarch64-linux-gnu/include/c++/11/bits/locale_classes.h:40,
                 from /usr/aarch64-linux-gnu/include/c++/11/bits/ios_base.h:41,
                 from /usr/aarch64-linux-gnu/include/c++/11/ios:42,
                 from /usr/aarch64-linux-gnu/include/c++/11/ostream:38,
                 from /usr/aarch64-linux-gnu/include/c++/11/iostream:39,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.hpp:31,
                 from /home/mashitian/project/Payload-SDK/samples/sample_c++/platform/linux/manifold3/application/application.cpp:26:
/usr/aarch64-linux-gnu/include/stdlib.h:153:8: error: ‘_Float128’ does not name a type; did you mean ‘_Float32x’?
  153 | extern _Float128 strtof128 (const char *__restrict __nptr,
      |        ^~~~~~~~~
      |        _Float32x
/usr/aarch64-linux-gnu/include/stdlib.h:246:25: error: ‘_Float128’ has not been declared
  246 |                         _Float128 __f)
      |                         ^~~~~~~~~
/usr/aarch64-linux-gnu/include/stdlib.h:331:8: error: ‘_Float128’ does not name a type; did you mean ‘_Float32x’?
  331 | extern _Float128 strtof128_l (const char *__restrict __nptr,
      |        ^~~~~~~~~
      |        _Float32x

mstsky115, 2025-09-03:
> [来自 Leon 在Zendesk 票证 #155270](https://djisdksupport.zendesk.com/agent/tickets/155270)中的代理评论：您好，您注意看下编译环境，从您的编译配置来看，使用的编译链工具是 aarch64-linux-gnu-g++，但是包含的头文件里有/usr/include/x86_64-linux-gnu/bits/floatn.h，表明是交叉编译时使用了这个和目标架构不一致的系统文件头，需要您设置正确的 sysroot。
> 
> _°°°_

https://developer.dji.com/doc/payload-sdk-tutorial/cn/manifold-quick-start/manifold-devlopment.html
是根据这个妙算开发环境教程配置的，我是小白，跟着你们的教程弄的，没见说要设置sysroot，不知道咋设置，系统也是Ubuntu 22.04
[image]

durka, 2025-10-16:
The cause seems to be the paths coming from pkg-config for ffmpeg. Commenting out the `pkg_check_modules(FFMPEG ...)` line in `manifold3/CMakeLists.txt` gets rid of the error.

Eaglewzw, 2025-11-24:
有人可以说一下怎么解决吗？原链接失效了，看不到了.......